prusa3d/PrusaSlicer #2401: Delete key no longer deletes objects (must use backspace instead)

timur-tabi, 2019-05-28:
### Version
2.0.0+

### Operating system type + version
Mac OS X 10.9

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
In Slic3rPE 1.x, the delete button could be used to delete objects.  Now I need to press the Backspace button instead.  This is counter-intutitve.

JohnOCFII, 2019-05-28:
Pressing the delete key in MacOS 10.14.5 with PrusaSlicer 2.0.0+ deletes objects from the main build area for me.  (Of course, on this 2018 MacBook Pro, there IS no key labeled backspace...)

bobstro, 2019-06-01:
On MacOS 10.14.5 Mojave on 2017 iMac 5K can confirm that selecting an object and pressing the [DEL] key has no effect. Must use the [BACKSPACE] key. It has been this way for many releases.

bubnikv, 2019-06-03:
We shall re-enable delete on OSX. There is no delete key on our OSX
laptops, that's why we removed it.

YuSanka, 2019-07-29:
Thanks for your report. This bug is fixed and will be a part of upcoming release.

closed 2019-07-29 by YuSanka