prusa3d/PrusaSlicer #2402: Change Port to Octoprint connection

AdronTech, 2019-05-28:
### Version
Version 2.0.0+linux64

### Operating system type + version
Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
CR-10 with Klipper and Octoprint

### Behavior
I use a reverse proxy on my server to access my printer and i mapped the reverse proxy to a port other than 80 or 443.

_Is this a new feature request?_
Would it be possible to add the feature to specify the port on which the printer listens. Eventually as a new field or in the hostname field after a colon.

vojtechkral, 2019-05-28:
> or in the hostname field after a colon

That's exactly how it works (it should, anyway). The field takes pretty much any URL syntax.

AdronTech, 2019-05-28:
@vojtechkral ok you are right. I vergott to add the http:// at the beginning.

closed 2019-05-28 by AdronTech