prusa3d/PrusaSlicer #2403: "Gtk-CRITICAL" messages on command line (PrusaSlicer 2.0.0, AppImage, Ubuntu 18.04)

simonkuehling, 2019-05-28:
### Version
PrusaSlicer 2.0.0+linux64 (AppImage)

### Operating system type + version
Ubuntu 18.04 64-bit

### Behavior
When launching PrusaSlicer from the command line, I get several of the following messages during use. The software is working just fine, no issue whatsoever - but I thought maybe this is useful information for the development team anyway...? are preferred_

```
(prusa-slicer:18908): Gtk-CRITICAL **: 09:22:34.936: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
```
```
(prusa-slicer:18908): Gtk-CRITICAL **: 09:22:34.964: IA__gtk_tree_row_reference_new_proxy: assertion 'path->depth > 0' failed
```

Let me know if I can provide more details on my system or do further testing if needed. If those messages are considered harmless or expected, feel free to close this issue!

vojtechkral, 2019-05-28:
Yeah, those messages are there. I don't think we've looked into them thoroughly, because they don't seem to cause any actual trouble, but most likely this is caused by wxWidgets one way or another. We don't use GTK directly.

simonkuehling, 2019-05-28:
ok, that's fine for me.. thanks for your quick response, I'll close this now...

closed 2019-05-28 by simonkuehling

bubnikv, 2019-06-03:
We should fix these asserts though. We went through a round of assert
fixing at the wxWidgets side, now it is time to eradicate these GTK assert
issues.

amatulic, 2019-09-29:
@simonkuehling and @bubnikv -- I won't reopen this, but I'm commenting to let you know that I'm getting many of these messages in PrusaSlicer2.1.

I get a dozen or so of these identical errors:
<code>(prusa-slicer:17129): Gtk-CRITICAL **: 07:53:00.044: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed</code>

...and four of these:
<code>(prusa-slicer:16708): GLib-GObject-WARNING **: 07:50:54.885: g_object_weak_unref: couldn't find weak ref 0x7fb9a7f00e00(0x3b60a68)</code>

...and two of these:
<code>** (prusa-slicer:16708): CRITICAL **: 07:50:55.250: gail_notebook_real_remove_gtk: assertion 'obj' failed</code>

joysfera, 2019-10-21:
FYI, I get frequent total unresponsiveness - mouse cursor freezes for couple of seconds every few seconds (gnome-flashback nvidia Ubuntu 18.04) and it seems related to these Gtk-CRITICAL messages - at least when the mouse pointer becomes responsible again for a moment (when it kinda unfreezes) then this message is printed in the console.