prusa3d/PrusaSlicer #2404: Object location on print surface is not saved/loaded correctly from *.3MF project

th33xitus, 2019-05-28:
### Version
2.0.0

### Operating system type + version
Win7

### Behavior
* _Describe the problem_
3MF file doesn't contain correct object location on print surface?

* _Steps needed to reproduce the problem_
Load multiple objects on print surface, rearrange them, save as 3MF, close program, open program, load 3MF file

* _Expected Results_
All objects on their respective locations where they were located

* _Actual Results_

Former location:
[image]

Object location the saved 3mf file got loaded after program restart:
[image]

bubnikv, 2019-06-03:
I believe this issue has been fixed with 0132313ee0cc6336eb6d40cf52db156d4f3f701b in the 2.0.0 final. 
(Fixed position of instances after paste object)

Do you get this issue with the 2.0.0 final as well? If you happened to load the 3MF file stored by 2.0.0-rc2 and older, you may see this issue in 2.0.0 final. If you create such a 3MF with 2.0.0 final, you should not be able to reproduce this bug anymore.

th33xitus, 2019-06-03:
You are correct! It's fixed in the 2.0.0 final.
I simply forgot to update my rc2 to the final release. My bad!

Closed

closed 2019-06-03 by th33xitus