prusa3d/PrusaSlicer #2406: Top layer sagging caused by auto cooling

githubcura35, 2019-05-28:
### Version
2.0

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Creality Ender 3

### Behavior
Top layer sagging caused by auto cooling. More cooling is needed for the top layer.

* _Steps needed to reproduce the problem_
  print the file in the project
* _Expected Results_
 smooth top layer
* _Actual Results_
 top layer is very glossy and there is sagging which shows the infill underneath

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

[PrusaSlicer_TR8x8_Support_BED.zip](https://github.com/prusa3d/PrusaSlicer/files/3228678/PrusaSlicer_TR8x8_Support_BED.zip)

githubcura35, 2019-05-28:
Possible related to #942 #378

bubnikv, 2019-08-01:
Do you have an example of another slicer that does a better job?

What you are experiencing is called pillowing. You need to increase the number of top layers.

githubcura35, 2019-08-09:
I had better results in Cura 4.0. I was suprised to find out that I had 8 top layers in my Cura profile.
I bumped up the number of top layers to 4 and fan min speed to 40% in PrusaSlicer. Now the results are much better in PrusaSlicer. Problem was between chair und keyboard as most of the time ;-)

closed 2019-08-09 by githubcura35