prusa3d/PrusaSlicer #2407: Prusa Slicer 2.0.0 mmu2 slicing issues

plikos, 2019-05-28:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

lukasmatena, 2019-05-30:
If there is a bug, we'll be grad if it gets reported. However, an "issue" like this will be closed soon.

danielegger1, 2019-05-30:
I have some slicing issues with a MMU2 print, there is a gap at the head which makes this model unable to print

AMF File attached
[Body_Black.zip.zip](https://github.com/prusa3d/PrusaSlicer/files/3237492/Body_Black.zip.zip)

Version PrusaSlicer 2.0.0
Model https://www.thingiverse.com/thing:2831169
Printer MK3S + MMU2S

[image]
[image]

lukasmatena, 2019-05-31:
Your model seems to be broken - fixing it with Netfabb solves the issue. PrusaSlicer just did not manage to detect or fix the issue. This is similar to https://github.com/prusa3d/PrusaSlicer/issues/1941, where you can find more detailed explanation. Hopefully we will improve this one day.

closed 2026-09-29 by prusaslicer-bot[bot]