prusa3d/PrusaSlicer #2408: Incorrect model volume calculation

camerriman, 2019-05-28:
### Version
2.0.0+

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

macOS Mojave 10.14.2
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

N/A
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

incorrect calculation of stl model volume
import stl
set expert mode
model info pane incorrectly displays the volume of the stl file. 
expected the correct volume  (168,675.3mm3)
significant incorrect volume calculation  (47047.20mm3)

** note - the previous version of slic3r worked perfectly :)
[image]
[image]
[image]

* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-06-03:
please attach your model/.

rtyr, 2021-01-29:
No response. Closing.

closed 2021-01-29 by rtyr