prusa3d/PrusaSlicer #2411: Color print issue in vase spiral mode

espr14, 2019-05-28:
v2.0.0 win64

[vase.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3229807/vase.3mf.zip)
Preview:
[image]

bubnikv, 2020-12-09:
What's wrong?

bubnikv, 2020-12-10:
What did you expect?

[image]

Printing a reasonable object works in spiral vase mode with color change, thanks @YuSanka for trying that.

[image]

Pane kolego, měl byste více přemýšlet, než se rozhodnete úkolovat jiné. Děkuji pěkně a přeji příjemně prožité vánoční svátky.

closed 2020-12-10 by bubnikv