prusa3d/PrusaSlicer #2413: Gcode misses infill lines but Preview looks correct

kingargyle, 2019-05-29:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

2.0.0

### Operating system type + version
_What OS are you using, and state any version #s_

Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Ender 3, Marlin 1.1.4

### Behavior
* _Describe the problem_

When trying to print after slicing the Upper Support.stl from the Imperial Terrain bundle.  When it is laying down the first layer, the rear left hand side of the layer is acting like it is under extruding or missing lines.   The same STL file, sliced with similar base settings in Cura does not have this issue.  It lays down the layers as expected.  Bed is level and there isn't a jam or extruder jamming or slippage.

* _Steps needed to reproduce the problem_

Obtain the the Upper Support.stl file from Imperial Terrain for the for the BundleFrontLines package.   Slice it with the base settings, and start printing it for the Ender 3.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_

Expected results is a good first layer that doesn't appear to be missing lines when infilling.

* _Actual Results_

Missing or skipped lines on the infill.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_

No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

[Upper Support.zip](https://github.com/prusa3d/PrusaSlicer/files/3230206/Upper.Support.zip)

kingargyle, 2019-05-29:
Here is a picture of the actual problem when it was printing. 

https://photos.app.goo.gl/wFzZZX7wmn7CSVxK7

The version that I did with Cura is printing fine currently.  Both the Red and Black were sliced and gcode generated with PrusaSlicer 2.0.0

Promite2501, 2019-05-29:
Looks like a few issues going on here.  possibly fixable by existing settings rather than a problem with the silcer.

Perhaps try reprinting with this settings bundle.  I think it will help your print. [https://www.facebook.com/groups/Ender3/permalink/684649551972138/](url)

Let us know how you go.

kingargyle, 2019-05-29:
Will take a look, pending approval to join that group.

kingargyle, 2019-05-30:
Downloaded the profile from the group, will try a print with the new profiles. Will post back with results.

kingargyle, 2019-05-30:
Closing this issue as the Ender3 .20 - Fast profile seems to be working.  At least I'm not seeing the same issue.  Will re-open if it continues to happen.

closed 2019-05-30 by kingargyle