prusa3d/PrusaSlicer #2415: Detect bridging perimeters: wrong flow ?

guestisp, 2019-05-29:
### Version
2.0.0

When slicing with `overhangs` set to one, some artefacts are added in the print. This is visible, in example, in the Benchy cockpit.

When `overhangs` is enabled, the overhang in the cockpick has a very very huge flow.
When setting `bridge_flow_ratio` to 1 I would expect the same flow as the rest of the print, thus the lines in the preview should be as thick as the other lines.

Nothing changes even lowering the bridge flow. The only way to get a line thickness similiar to the rest of the print is to set `bridge_flow_ratio` to 0.25 or similiar. This should be 25% of the default flow.

Is this expected ? I would expect that in a default configuration, a bridge would be handled normally, with the same flow as the rest of print.

[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/3231625/3DBenchy.zip)

bubnikv, 2019-06-03:
The bridging flow extruding a rounded thread of the nozzle diameter cross
section. If you enable the "overhangs" feature, the overhang perimeters are
extruded as bridges. If you don't like it, you may disable this feature.
Ideally, the excess extrusion would be higher for less supported extrusions
and lower for more supported extrusions.

guestisp, 2019-06-07:
I like the idea, but seems wrong to me to overextrude a bridge. Currently, is overextruding (a lot)...... Look at the benchy, this defect is also visible in the preview.

As the extrusion multiplier for bridges/overhangs is set to 1, i would expect that the same flow as everything else would be used. Currently, to have (more or less) the same flow, I have to set the value to .25

rtyr, 2021-01-29:
Answered. Closing.

closed 2021-01-29 by rtyr