prusa3d/PrusaSlicer #2417: Bad Allocation error

pfukan, 2019-05-29:
### Version
PrusaSlicer 2.0.0+win64

### Operating system type + version
Windows 10 Professional 1809, 64-bit

### 3D printer brand / version + firmware version (if known)
i3 MK3S+MMU2S stock model / firmware 3.7.1 printer + 1.0.5 MMU

### Behavior
Slicing model with soluble supports. In phase of exporting g-code there is an "bad allocation" error raised and no g-code is exported.
When I use different model placement using the place on face function it is sliced and exported properly.
What's wrong with the original model placement?

#### Project File (.3MF) where problem occurs

[pegasus_tg_error.zip](https://github.com/prusa3d/PrusaSlicer/files/3232687/pegasus_tg_error.zip)
[image]

bubnikv, 2019-06-03:
How much physical memory do you have on your computer?

pfukan, 2019-06-03:
I have 16 GB RAM on my computer. Just now after slicing another multimaterial model there is 6,6 GB of memory occupied and 9,2 GB still available.

bujocek, 2019-07-28:
I have the same problem. Mostly it fails, but sometimes after tweaking settings it finishes successfully. It has to be some error in support generation because in my project file it is enough to play with the "Pattern spacing" parameter in Support options to get it working. I.e. when I set "Pattern spacing" to 2.8mm than the gcode is generated fine but when I lower it to 2.5mm or raise it to 3mm than it fails on "bad allocation". Here is my project file:
[stream_fail.zip](https://github.com/prusa3d/PrusaSlicer/files/3438943/stream_fail.zip)

bubnikv, 2019-07-31:
Please check whether your issue is not the same as of #2682 
That is, you may try to select a different infill from Gyroid, and you may disable the silent mode time estimates.

bujocek, 2019-07-31:
I have rectilinear infill, and switching off silent mode does not help. See the project file attached in previous post. I have 7.9GB free ram (out of 16GB) and the PrusaSlicer fails really quickly - it doesnt look like it runs out of memory during the gcode generation. I tried it on two different computers and the problem stays. It is easily reproducible so it should be possible to research by dev.

lukasmatena, 2019-08-05:
The issue could be caused by an access violation bug in wipe tower generator, same as reported in https://github.com/prusa3d/PrusaSlicer/issues/2581. This bug should be fixed in master now. I myself never got a bad_alloc exception on my computer, but outcomes of this kind of error are generally unpredictable.

@pfukan, @bujocek
Could you guys please retest in PrusaSlicer2.1.0-alpha1 (which will be released shortly) and let us know if it is ok for you? I believe it will be, but I'd like to be sure. Thanks.

pfukan, 2019-08-05:
Tested in PrusaSlicer2.1.0-alpha0 (commit 6da196b). My problematic 3mf projects were sliced without error with this version.

bujocek, 2019-08-06:
I see only a tag with sources. Is there maybe a windows binary to test? I don't have Visual studio 2019 to build the whole project myself.

bubnikv, 2019-08-07:
@bujocek please be patient.

bubnikv, 2019-08-24:
@pfukan thanks for testing. Closing.

closed 2019-08-24 by bubnikv

HaasNL2, 2019-09-28:
Had the same error, went away when I switched to another simpler infill from the 3D gyro infill.