prusa3d/PrusaSlicer #2418: MMU2 Slicer temperature

neddiggis, 2019-05-29:
### Version
_2.0.0+win64_

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2s - Firmware 3.7.1

### Behavior
When priting with 2 different materials (PLA and PTEG) the temperature changes before the unload, meaning the PTEG is unloading too cold and the PLA is unloading too hot. The Temp change should happen while the filament is being retracted into the MMU.

_Is this a new feature request?_

No, it's a bug.

[mmu2_PLA2PET5-40-20_2h14m.zip](https://github.com/prusa3d/PrusaSlicer/files/3233494/mmu2_PLA2PET5-40-20_2h14m.zip)

chrisboer, 2019-06-15:
I have the same issue but even with single material on MMU2S. 
So in slicer I set PLA first layer temp 215 and other layers 215
The printer lowers the temperature to PLA Prusa default 205 after first layer.
**;AFTER_LAYER_CHANGE
;0.4
M104 S205 ; set temperature**

Why is this put in when I did not ask for this to be put in... where can I remove these default.

M104 S215 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp

johnnyruz, 2019-07-03:
@neddiggis I had this exact same issue when printing with PLA and PVA soluble supports. PLA is set to 205 and the PVA is set to 195. The decrease in temperature during the PLA cooling moves ruined the tip on the PLA causing failures, and the increase in temperature during the PVA cooling moves caused all sorts of problems as well.

I agree it makes much more sense to send the temperature change code just before the 'T' command to unload/load the filament.

As a test I wrote a post-processor script that moves the temp change lines around and it completely fixed my issues. Just finished an 8 hour, 154 tool change print with soluble interface supports without any errors.

neddiggis, 2019-07-03:
@johnnyruz that's great news. Any chance you could share that script?

johnnyruz, 2019-07-03:
@neddiggis I just created a github repo with the script attached. I just updated the Print Settings > Output Options to have Prusa Slicer call that script.

https://github.com/johnnyruz/PrusaScripts

**Use at your own risk!!** I've used it successfully for multiple prints without any issues but that doesn't mean there's not some bug or a different setting in your configuration that would screw up the pattern detection in the file!

neddiggis, 2019-07-03:
@johnnyruz thanks! I've given it ago and it's not running. Do I need a specific version of Python?

6Edx, 2019-08-05:
@johnnyruz

Hello, I tried running this script and it gives me an error code 1. It seems there is somme issues with the readlines(). i tried to juste read and print out the data and that fails also. You have any clue what i  could be doing wrong?

m3cinc, 2019-08-06:
The need for controlled temperature during integral time of retraction and wipe is necessary to prevent clogging when 2 or more materials are used. I face the same issue using PVA as soluble support in a PETG part. Currently the PETG left over in the hotend must cool below its melting point and clogs the extruder. The PVA is inserted into the hotend above its upper temperature limit and degrades... This temperature handling must be resolved to enable true multi-material printing. Not to mention the management of the filament tip... Tha is another story altogether!

cggorman, 2019-08-06:
I agree, but in the mean time you may want to consider using a different support. Perhaps Airwolf Hydrofill, BVOH, HIPS, maybe even PLA (sodium hydroxide soluble, careful)?

6Edx, 2019-08-07:
@m3cinc I think the temperature of the previous filament must be kept until the filament is unloaded. Then the hottest of both temperature is used for half a sequence of the purge and changed to the current filament temperature to finish the other half of the purge sequence.
This way you purge the pottentiel clogs and the degraded filament is also purged and the tips wont be affected by the wrong temperature.

@cggorman of course buying new materials with the same range of temp is a solution, but i love my petg... hahahaha and my process needs food grade materials even is the food isnt in contact with the material.

6Edx, 2019-08-07:
Seems this issues has been adressed.
https://github.com/prusa3d/PrusaSlicer/issues/1339
i will test it out and give some feed back

closed 2026-09-29 by prusaslicer-bot[bot]