prusa3d/PrusaSlicer #2420: Suppress/Disable default start gcode

Rudgas, 2019-05-29:
### Version
2.0.0

### Operating system type + version
Linux 5.1.3-arch2-1-ARCH Arch Linux

### 3D printer brand / version + firmware version (if known)
Hypercube Evolution, Marlin bugfix (current)

### Behavior

Prusa-Slicer is generating it's custom start-gcode for every file and while some gcode is appreciated, (heat bed, heat nozzle) other not so much, i.e. retracts and unretracts right before the beginnning.

`; generated by PrusaSlicer 2.0.0 on 2019-05-28 at 20:25:12

; external perimeters extrusion width = 0.48mm
; perimeters extrusion width = 0.48mm
; infill extrusion width = 0.48mm
; solid infill extrusion width = 0.48mm
; top infill extrusion width = 0.48mm
; support material extrusion width = 0.48mm

M201 X1000 Y1000 Z100 E1000 ; sets maximum accelerations, mm/sec^2
M203 X150 Y150 Z3 E30 ; sets maximum feedrates, mm/sec
M204 P1000 R1000 T1000 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.30 E5.00 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107 ; disable fan
M190 S60 ; set bed temperature and wait for it to be reached
M104 S215 ; set temperature
M900 K0
G28
M420 S1 ;enable automatic bed leveling
; Filament gcode
M109 S215 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0 ; reset extrusion distance
G1 Z0.300 F9000.000 ; move to next layer (0)
G1 E-4.00000 F2400.00000 ; retract <----- this I want to avoid
G92 E0 ; reset extrusion distance
G1 Z0.500 F9000.000 ; lift Z 
G1 X108.265 Y104.441 ; move to first skirt point 
G1 Z0.300 ; restore layer Z 
G1 E4.00000 F2400.00000 ; unretract <----- this I want to avoid
G1 F2100 <--- Actual printing starts from here
G1 X109.387 Y103.287 E4.08344 ; skirt`

I see no reason to do a reatract/unretract move right at the beginning, the printer was heating the nozzle right before, and only if a custom start gcode for priming was inserted before this block, will the retract be somewhat useful. Chances are good a retraction was performed at the end of the last print, so this retract/unretract will have a good chance of clogging the nozzle at the beginning retracting warmed up filament. 
Expected behavior is a priming of the nozzle via the skirt, so this retract/unretract is better placed after the skirt (if at all).

I request removal of this retract/unretract move or better yet, the ability to completely disable the auto-generated startup gcode (expert option).

#### Project File (.3MF) where problem occurs
All files.

neophyl, 2019-05-31:
If you don't like the start gcode just set up a second printer profile with the gcode that you do want to use.  I cant see the point of having a tickbox to disable it as its just as quick and easy to select a different profile.

Rudgas, 2019-05-31:
I totally agree, but how? I have some custom G-code as start-code as you can see in the code above:
```
M900 K0
G28
M420 S1 ;enable automatic bed leveling
```
I also tried adding a G21 to the filament-specific G-code, but it doesn't help either, retract/unretract is still there as before as well as all the other auto-generated code.

neophyl, 2019-05-31:
I am not sure that I am understanding your difficulties.  Please excuse if you already know any of the following.

PrusaSlicer like the slicer its based on allows you to create multiple profiles for 3 areas, the Print Settings, the Filament Settings and the Printer Settings.  

I am sure you are familiar with the Print and Filament settings profiles as there are several pre installed examples to get you started (0.20mm QUALITY MK£ for example on print settings or Prusa PLA on filament settings.  You can also create multiple PRINTER profiles under Printer Settings.  
PrusaSlicer doesnt allow you to modify the built in profiles but you can modify them and then save as a new profile which you can then just select via the drop downs as usual.

Custom G code is added in 2 locations.  It can be added under Custom GCode as part of a Filament profile.  The filament profiles usually only have the M900 command in them by default though.

The second place and the one used normally is under Custom G code located in the Printer Profile.  This has 4 sections Start gcode, end gcode, Before Layer change and after layer change.
This where all the custom code is placed that handles bed mesh leveling, the initial purge line etc.  

You can edit this to your hearts content and save as a new profile.  The Printer Profile also has all your other printer specific settings in there. 

I'm not sure where those unretract ones in your listing are coming from, here is the start of one of my files

; generated by PrusaSlicer 2.0.0+win64 on 2019-05-29 at 15:05:03
;
; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.35mm
; first layer extrusion width = 0.42mm

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.35mm
; first layer extrusion width = 0.42mm

M73 P0 R751
M73 Q0 S776
M201 X1000 Y1000 Z1000 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E1.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M115 U3.6.0 ; tell printer latest fw version ; <--- this is the the start gcode from the printer profile
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S210 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S210 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0 F1000.0 ; intro line
M73 Q0 S776
M73 P0 R751
G1 X100.0 E12.5 F1000.0 ; intro line
G92 E0.0
M221 S95 <--- this is the end of the start gcode from the printer profile
M900 K30; Filament gcode ,<--- this line is from the custom gcode under Printer settings, it only has 1 line in there.
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2

G1 E-0.80000 F2100.00000
G1 Z0.600 F10800.000
;AFTER_LAYER_CHANGE
;0.2
G1 X85.491 Y51.016
G1 Z0.200
G1 E0.80000 F2100.00000
M204 S1000
G1 F1200 <--start of print here
G1 X86.395 Y50.796 E0.02916 

As you can see I don't have the un-retracts in there.

Rudgas, 2019-06-01:
Thanks for the detailed explanation! I think I have indeed found the culprit.

For clarification: I created my own printer-profile from scratch and adjusted it for my hypercube printer. So no presets are included automatically in the settings. Settings are also in expert mode, so I don't miss anything hidden.
Now for comparison, I added the prusa i3 mk3s profile just to check out the differences. As I now believe, the re/unretract part of the gcode is part of the layer-change-gcode.
As you can see, in your code the re/unretract is also present, with 0.8mm as is the amount of retraction you set in the printer-tab settings. (For my printer it's 4mm so that's why it says 4 in my gcode).
```
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2

G1 E-0.80000 F2100.00000
G1 Z0.600 F10800.000
;AFTER_LAYER_CHANGE
;0.2
G1 X85.491 Y51.016
G1 Z0.200
G1 E0.80000 F2100.00000
```
However, for your/i3 gcode, this re/unretract actually makes sense as a nozzle-priming code is included right above this and the printer moves a primed nozzle to the beginning of the print. But this same re/unretract is still inculded if the printer is not a prusa and the "prusa-priming" start-gcode is not included. Thus doing the re/unretract with an unprimed nozzle :-)

Here are three sample-codes from my settings to show how and where the re/unretract is happening:

This is with retract on layer change disabled:
```
G92 E0 ; reset extrusion distance
;here the layer change gcode starts
G1 E0; <-- I added this
G1 Z0.100 F9000.000 ; move to next layer (0)
G1 E0; <-- I added this
;here the layer change gcode ends
G1 E-4.00000 F1800.00000 ; retract
G92 E0 ; reset extrusion distance
G1 Z0.300 F9000.000 ; lift Z
G1 X120.983 Y107.730 ; move to first skirt point
G1 Z0.100 ; restore layer Z
G1 E4.00000 F1800.00000 ; unretract
M204 S600 ; adjust acceleration
G1 F2100
G1 X173.219 Y107.680 E4.90895 ; skirt 
```

This is with retract on layer change enabled:
```
G92 E0 ; reset extrusion distance
;here the layer change gcode starts
G1 E0; <-- I added this
G1 E-4.00000 F1800.00000 ; retract
G92 E0 ; reset extrusion distance
G1 Z0.200 F9000.000 ; lift Z
G1 E0; <-- I added this
;here the layer change gcode ends
G1 X120.983 Y107.730 ; move to first skirt point
G1 Z0.100 ; restore layer Z
G1 E4.00000 F1800.00000 ; unretract
M204 S600 ; adjust acceleration
G1 F2100
G1 X173.219 Y107.680 E4.90895 ; skirt
```

And this is with retraction disabled (=0):
```
G92 E0 ; reset extrusion distance
;here the layer change gcode starts
G1 E0; <-- I added this
G1 Z0.100 F9000.000 ; move to next layer (0)
G1 E0; <-- I added this
;here the layer change gcode ends
G1 X120.983 Y107.730 ; move to first skirt point
M204 S600 ; adjust acceleration
G1 F2100
G1 X173.219 Y107.680 E0.90895 ; skirt
```

CGuasco, 2019-07-24:
I Have the same problem. 
I DON'T WANT M201, M202, M203, M204, M205.
In particular M204 Rxxxx create BIG BIG Problems at MarlinKimbra firmware,
It revert Z movement and immediateli crash Z axis.
The creator of marlinkimbra say that the M204 command has wrong parameters and dont't explain why M204 R1500 revert Z axis.
In order to use PrusaSlicer i needed to comment M204 parsing in firmware.
IS NOT THE CORRECT WAY, only yust to work.
In all profiles i found custom GCode, but no one insert M204 and others.
appear that are inserted directly from software (according to me is a big error)
I leaved only 1 printer profile, that contains codes as G28; G1 Z0.5 F1000 etc 
BUT NO CODE M201,202,203,204,205 etc.
Where are these codes, so I can delete! 
Many thanks to all.
Regards 
Carlo

StephenRC, 2019-07-24:
One way to have them in the gcode, is to tell the slicer you're using
RepRap/Sprinter firmware.  

On Wed, 24 Jul 2019 09:46:26 -0700, CGuasco <notifications@github.com>
had a flock of green cheek conures squawk out:

> I Have the same problem.
> I DON'T WANT M201, M202, M203, M204, M205.
> In particular M204 Rxxxx create BIG BIG Problems at MarlinKimbra firmware,
> It revert Z movement and immediateli crash Z axis.
> The [...]

CGuasco, 2019-07-25:
Ok, 
Actually using RepRap / Sprinter instead of Marlin there are no more strange codes.
I hope there are no contraindications to using RepRap / Sprinter setup with Marlin firmware.
Another way I was trying is to write a postprocess  (in print setup) that deletes the unwanted lines.
Thanks for your advice
Carlo
> 
> One way to have them in the gcode, is to tell the slicer you're using RepRap/Sprinter firmware. On Wed, 24 Jul 2019 09:46:26 -0700, CGuasco <notifications@github.com> had a flock of green cheek conures squawk out:
> […](#)

bubnikv, 2019-08-01:
Could you close this issue then?

CGuasco, 2019-08-01:
Resolved.
Using postprocess.
There is not any option "close issue"

Rudgas, 2019-08-01:
This issue is most certainly not closed.

(Manual) Postprocessing to remove erroneous gcode is neither good practice nor a solution. Neither is changing the target firmware in an attempt to suppress it with unpredictable/unforseeable consequences for the target firmware and it's custom commands.

CGuasco, 2019-08-01:
An other solution is to use Cura instead of Slic3r/ or prusaslicer.
I don't found an old version of slic3r,  so there are two solutions... 1) Postprocessing, 2) Cura .

lukasmatena, 2019-08-01:
@CGuasco And the third solution is to not set Marlin gcode flavor, which you yourself mentioned in your previous post. Also, you don't have the "same problem" as @Rudgas, you are the first one who mentioned anything about the M204 etc. commands. The original issue was about retractions.

@Rudgas I'm sorry but I don't understand what's "erroneous". The M204 etc. are standard Marlin gcodes. If you're running Marlin, your printer should recognize them and there is no problem. If you're not running Marlin-conforming firmware , don't set gcode flavor to Marlin. Then there is also no problem.

As for the retractions, I also don't understand. I'm reading the gcodes you posted and all seem fine to me. Whether or not there is priming in custom start gcode, it makes sense to me to retract for the travel move to the print and unretract there.

> Chances are good a retraction was performed at the end of the last print

No, chances are that the user has switched filaments before the print. And that's also much better defined initial state than "some retraction might have happened".

> so this retract/unretract will have a good chance of clogging the nozzle at the beginning retracting warmed up filament

There are thousands of retractions during a print, what is so special about this one that it will clog the nozzle? If you're worried, turn retractions off - it will disable this retraction as well, as you have demonstrated.

bubnikv, 2019-08-02:
@lukasmatena
I did not have time & patience (there are just too many issues open) to
carefully analyze. It is true, that if the melted filament is retracted
twice or too far, it may expand and clog inside the heatbreak.

brew99, 2019-10-02:
I'm also wondering if this (supression of M201, M203, M204, etc) can be a simple check/uncheck box in "Printer Settings" - "Machine limits"?   It would then allow default marlin values to be used rather than what is in the slicer.  I'm using v2.1 final and these M codes are added to the .gcode file, with no way to disable them without editting the .gcode file

Jebtrix, 2019-10-02:
@brew99 
> with no way to disable them without editting the .gcode file

Technically you could do a M501 (Load all saved settings from EEPROM) at beginning of custom Start G-code. This will effectively undo the M2xx.

daleckystepan, 2019-10-02:
@Jebtrix "Use firmware acceleration/jerk" checkbox seems to be more consistent and transparent solution.

Jebtrix, 2019-10-02:
@daleckystepan Agreed. I'm just saying its possible to workaround without editing resulting .gcode file.

daleckystepan, 2019-10-02:
Very easy workaround is also to set G-code flavor to Reprap/Sprinter. It generates exactly same G-code but no M201, M203, M204 commands.

Jebtrix, 2019-10-02:
I've never looked into differences of flavor output from prusaslicer. I'll take your word for it.

rjacobs1969, 2020-01-17:
>I DON'T WANT M201, M202, M203, M204, M205.

>Very easy workaround is also to set G-code flavor to Reprap/Sprinter. It generates exactly same G-code but no M201, M203, M204 commands.

Thank god for this thread, exactly what I was looking for. 
Setting to  G-code flavor to Reprap/Sprinter worked, though IMO it shouldn't add the M20x commands when using  Marlin as setting.

jrezin, 2020-04-02:
I'm having a similar problem.

I've defined my _START_G-CODE_ as :

```
FIRST_LINE_START [first_layer_bed_temperature]
SECOND_LINE_START [first_layer_temperature]
```

but no matter how I change configs, inis and etc, I end up with this _START_G-CODE_ :

```
M73 P0 R15
M107
M190 [first_layer_bed_temperature] ; set bed temperature and wait for it to be reached
M104 [first_layer_temperature] ; set temperature
FIRST_LINE_START [first_layer_bed_temperature]
SECOND_LINE_START [first_layer_temperature]
```

The thing is, I've NOT set the _START_G-CODE_ as the resulting file.
Also, It's messing up my prints! I don't want it setting the temperature of the extruder (M104) at this time.
I also don't mind the M20X codes with the Marlin gcode flavor, but anyway, the START_G-CODE should be the actual _START_G-CODE_ I've set, not something else... especially if you can't fix it without manually editing it every single time.

I know it's not like you guys are forcing those changes on users, trying to be incompatible, but IMHO the resulting behavior is unwanted and, not only unclear to users, confusing and frustrating.

lukasmatena, 2020-04-03:
I don't understand what your FIRST_LINE_START will do, but if you want to suppress the M190 and M104 commands, use them in your custom gcode. So if you prepend you custom gcode with
```
M104 S0
M190 S0
```
PrusaSlicer will then not set the temperature for you.

> the resulting behavior is unwanted and, not only unclear to users, confusing and frustrating

I don't think that setting temperature is unclear or confusing and I would assume slicer to do it for me.

Wireheadbe, 2020-07-07:
Thanks for that suggestion. Those custom macro's are used by people using Klipper (I use it to) - your suggestion to set both temperatures to 0 first works. A checkbox "Really don't prefix anything before this start G-Code" would be nice...
My start macro is e.g. this: START_PRINT EXTRUDER_TEMP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature]
This in turn will first Home, then move the extruder with bed level sensor to the middle of the bed, heat the bed, so the probe heats up a bit as well. It then does a mesh map before printing, then going back zo X- Y0 Z0; and only then heat the extruder, as to not have any filament oozing on the bed. You can understand, if anything is prefixed before this code, probing around the bed might leave filament all over the place.

n8bot, 2020-07-11:
The user should be able to control all elements of the gcode file. It seems silly how much slic3r makes these decisions for users. The gcode files slic3r (and forks) generates are quite cluttered and messy. 

Every single line that is automatically inserted should be able to be suppressed. 

Why should we get so much control over so many slicing parameters, yet we can’t at all configure how we want the gcode?

CGuasco, 2020-07-12:
> The user should be able to control all elements of the gcode file. It seems silly how much slic3r makes these decisions for users. The gcode files slic3r (and forks) generates are quite cluttered [...]

Absolutely right,
All added gcode lines should be in an editable file. It is completely wrong to assume that everyone has an Prusa with Marlin 2.
There are other printers and other firmware. Also very popular (e.g. MK4duo, but not only)
I understand giving pre-configured profiles for prusa I3 ver 4.15 with Marlin 2.0.3 January 2020 edition.
But a product like Slic3r aka PrusaSlicer should continue to be open and configurable. What is worth criticizing the DaVinci if you then make the same mistake.

lukasmatena, 2020-07-12:
>  It is completely wrong to assume that everyone has an Prusa with Marlin 2

Nobody assumes that. There is a selection of several gcode flavors in Printer Settings to support many other variations. It takes  quite a lot of effort to keep it working by the way. And this work is rarely done by implementers of those firmwares (although it could be - it is "open and configurable" enough). PrusaResearch has of course no resources to do proper testing of all the other firmwares, but accusing us of not being open enough is quite hard for me to listen to.

n8bot, 2020-07-12:
The reprap flavour is not correctly specified. 

As a singular example, M601 is not supported by rrf, yet that code does not change when reprap flavour is selected. I know there are other cases too. M204 is not exactly correctly used. This is just off the top of my head.

CGuasco, 2020-07-14:
> Nobody assumes that. There is a selection of several gcode flavors in Printer Settings to support many other variations. It takes quite a lot of effort to keep it working by the way. And this work [...]
----- ----- -----
From 2014 i used Slic3R as the best slicer.
Now i use Simplify3d, is not free and have less performance....... but work.
Unfortunately, if, for example you have MK4duo, there are line codes that crash the printer (M204 for example, is used out of Standard parameters (https://reprap.org/wiki/G-code) ).
So I need to implement a post process script to eliminate M204, I don't want to edit EVERY gcode generated to avoid PHISICAL DAMAGE to my printer.
Finally i preferred to modifiy MK4duo to ignore M204 and other wrong gcode lines. Is not so flexible as you say.
When is so simple to implement the fixed coded lines in a text/xml file or in Personalized GCode  block, like Start GCode, End GCode, Before layer change etc etc. So the user can eliminate unuseful or wrong lines.
This require a lot less effort than create a WORKING setup for every firmware.

foreachthing, 2020-07-14:
> So I need to implement a post process script to eliminate M204,...

This should get you started:
http://projects.ttlexceeded.com/3dprinting_prusaslicer_post-processing.html

You'd just need to change line 27 (to find your M204 code) and line 30 (to comment it or simply remove it).

CGuasco, 2020-07-15:
> 
> 
> > So I need to implement a post process script to eliminate M204,...
> 
> This should get you started:
> http://projects.ttlexceeded.com/3dprinting_prusaslicer_post-processing.html
> 
> You'd just need to change line 27 (to find your M204 code) and line 30 (to comment it or simply remove it).

Thanks, I had written a simpler thing, but this procedure seems well written.
However, by now I have abandoned Slic3R / PrusaSlicer, for Simplify3D

n8bot, 2020-08-09:
I notice that the ability to customize color change and pause print G-Code as been added. Thank you very much for that!
[image]

rtyr, 2020-10-20:
It is now possible to disable emitting of the M20x commands to the g-code (since 2.3.0-alpha1 release).

[image]

leewaa, 2021-02-02:
This is still a persisting issue. Running the latest version it still seems to be introducing GCode *before* the configured *start* gcode. IMO this should not be happening. Also the inserted GCode is unconfigurable;. For example I need to first select the right tool before the starting the heat up. A

```
; generated by PrusaSlicer 2.3.0+ on 2021-02-02 at 14:53:44 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M107
M190 S60 ; set bed temperature and wait for it to be reached
M104 S200 ; set temperature
;TYPE:Custom
;Start

T1;

M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
M109 S200 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
```

leewaa, 2021-02-02:
Adding the temperature warm up to my start GCode resolved this.... 

The bug however still remains that if its not configured then `M104` is added unknowingly, hence the printer does not wait for the temperature to be reached.

rtyr, 2021-02-02:
PrusaSlicer will add M190/M104/M109 automatically if those commands are not detected in your start g-code.

rtyr, 2021-02-02:
Yes, if you specify only M104, then M109 is not automatically added. But I don't see any reason to make start g-code like that, it is basically user error.

leewaa, 2021-02-02:
Not user error - Sorry just to clarify @rtyr, I **did not** add `M104`, PrusaSlicer did - hence I faced this issue that the printer didn't wait for it to heat up.

lukasmatena, 2021-02-02:
@leewaa 
It does what it says:

[image]

leewaa, 2021-02-02:
> @leewaa
> It does what it says:
> 
> [image]

@lukasmatena where did you get that information ? It should be made clearer to the user.
**However** I don't see how that mean I'm wrong. I never Said that I inserted `M104` - Since https://github.com/prusa3d/PrusaSlicer/issues/2420#issuecomment-771707465 I added my own **correct* Gcodes for heating up the hotend - However **not** `M104.

I think you guys are miss understanding the problem here, I will try to be clearer.

- The GCode I posted in my initial https://github.com/prusa3d/PrusaSlicer/issues/2420#issuecomment-771695206 is what the software produces. 
- Everything before `; Start` is inserted without the user ever knowing about it. 
- So As you can see the **software inserts temperature setting for the hotend with `M104`**

After `; Start` - that is my own configured starting GCode. This is bad, the printer will start trying to print right away before the hotend as even reached temprature. Hence  `Software error`, not `User error`.

This could be serious problem for people using firmware that does not protect cold extrusion - it also wastes peoples time. 

My suggestion - regardless of what is coming in with the option to disable it - change that automatically inserted hunk from:
- `M104 S200 ; set temperature`
to
- `M109 S200 ; set hotend temperature and wait for it to be reached

leewaa, 2021-02-02:
Oh I see, ok sorry about that.

I do feel that should be made much clearer. I did not leave my curser hovered long enough for that Tool Tip to show up - hence I was left in the dark about it. Think about more "advanced/experienced" users pasting in their custom Start Gcode in there, not even giving that tool tip a chance to show up. I am sure I am not the only one who has run into this issue. People coming from other Slicers will also get confused by this. - So displaying it in a clear way would prevent confusion or user error such as this.

I know another matter - but just some feedback - This way of inserting the GCode can get confusing and without that piece of information - leads to user error. I see the point about it but some people have their own pre-print line or filament extrusion process. This is also not possible with that set up.

lukasmatena, 2021-02-03:
> Think about more "advanced/experienced" users

Advanced users know the software (by definition).

> I am sure I am not the only one who has run into this issue.

You're not. People sometimes run into this issue when assuming that their use case is the only one and therefore should be implicit. See e.g. https://github.com/prusa3d/PrusaSlicer/issues/3033#issuecomment-539893086.

> some people have their own pre-print line or filament extrusion process. This is also not possible with that set up.

Some people? All single extruder Prusa system profiles have a purge line in start gcode, why would it not be possible? Just use `M109 S[first_layer_temperature]` as explained in that tooltip and this thread multiple times.

> change that automatically inserted hunk to M109 S200 ; set hotend temperature and wait for it to be reached

Thanks. People would complain that it oozes during mesh bed levelling and how it is outrageous that PrusaSlicer does decisions for them. The discussion you joined was not about M104 but about lack of control. It is also a change that would break existing profiles.

> I do feel that should be made much clearer.

You might be right here. What do you suggest?

leewaa, 2021-02-03:
> Advanced users know the software (by definition).

Sorry, I meant people who are fair more advanced in the 3D printing world - not with PrusaSlicer.

> You're not. People sometimes run into this issue when assuming that their use case is the only one and therefore should be implicit. See e.g. #3033 (comment).

Ignorance is bliss 😂 😂 - assumptions are very easy to make specially when coming from other Slicer software.

> Thanks. People would complain that it oozes during mesh bed levelling and how it is outrageous that PrusaSlicer does decisions for them. The discussion you joined was not about M104 but about lack of control. It is also a change that would break existing profiles.

I hadn't considered this since I do not run mesh bed compensation before every print - though the complaints are understandable.

Put the description for each inline to the right so that it is always visible, perhaps with the option to hide it ?
I can't imagine GCode needing so much width.

[image]

Rudis1261, 2021-05-22:
Yeah it should be ok if you when you paste custom GCODE that it doesn't prepend from the default profile. Or at least have a checkbox or something to ensure that only the GCODE I paste in there is run. 

Maybe something like what Simplify 3D does, where those set nozzle and bed, and wait for them to heat to be settings in the printer profile, or printer settings and setting the GCODE based on that, not the custom GCODE section.

besser, 2021-10-20:
SuperSlicer has this option:

[image]

JRHeaton, 2022-07-18:
> SuperSlicer

We still need this in PrusaSlicer, 6+ months later. I don't need or want the slicer to insert invisible, non-explicit gcodes before my actual start gcode. This is a huge problem because I now don't actually have control over the start of my print.

pbsuper, 2022-07-22:
I really wanna try Prusaslicer with the new Arachne slicing feature but this is just stopping me from being able to use it for my machines without having to do extra work.

This feature needs to be fixed.

Sonnenwerk, 2022-07-25:
> This feature needs to be fixed.

I second this !

Slavfot, 2022-07-28:
> I really wanna try Prusaslicer with the new Arachne slicing feature but this is just stopping me from being able to use it for my machines without having to do extra work.
This feature needs to be fixed.

Agree. Those extra lines of code is ruining my klipper start gcode. I don't want to manually edit the gcode erasing extra lines before each print.
A checkbox like in superslicer would be a nice solution.

JasonRashall, 2022-08-03:
> Those extra lines of code is ruining my klipper start gcode. 

Support for this reason.

Biz810, 2022-09-26:
> SuperSlicer has this option:
> 
> [image]

Hi,
is there anything new about this problem with the start gcode? I've tried everything to solve this problem but nothing worked for me. As long as Prusaslicer doesn't take over this function I'll stay with Superslicer. It's actually a shame because I find Prusaslicer easier to use.

Slavfot, 2022-09-26:
> 
> Hi, is there anything new about this problem with the start gcode? I've tried everything to solve this problem but nothing worked for me. As long as Prusaslicer doesn't take over this function I'll stay with Superslicer. It's actually a shame because I find Prusaslicer easier to use.

You can get around this by adding the values that prusaslicer would add automaticaly in the custom start g-code and set them to zero.
Your custom Start G-code in prusaslicer could look something like this:

```
M104 S0 ; set extruder temp to zero
M140 S0 ; set bed temp to zero
M190 S0 ; wait for bed temp to zero
M109 S0 ; wait for extruder temp to zero

START_PRINT EXTRUDER={first_layer_temperature[initial_extruder]} BED=[first_layer_bed_temperature] FILAMENT=[filament_type]
```

This would result in a gcode looking like this when sliced:
```
M107 ;turn fan off (i have not found a way to get rid of this but i can't see this do any damage)
;TYPE:Custom
M104 S0 ; set extruder temp
M140 S0 ; set bed temp
M190 S0 ; wait for bed temp
M109 S0 ; wait for extruder temp

START_PRINT EXTRUDER=215 BED=60 FILAMENT=PLA
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion (this is toggleable in prusa slicer)
```
Then the rest could be handled by your START_PRINT macro.
And make sure that G21 G90 and M83 that is printed automatically after START_PRINT doesn't ruin anything for you.

Biz810, 2022-09-27:
> > Hallo, gibt es etwas Neues zu diesem Problem mit dem Start-Gcode? Ich habe alles versucht, um dieses Problem zu lösen, aber nichts hat bei mir funktioniert. Solange Prusaslicer diese Funktion nicht übernimmt bleibe ich bei Superslicer. Es ist eigentlich eine Schande, weil ich Prusaslicer einfacher zu verwenden finde.
> 
> Sie können dies umgehen, indem Sie die Werte hinzufügen, die Prusaslicer automatisch im benutzerdefinierten Start-G-Code hinzufügen und auf Null setzen würde. Ihr benutzerdefinierter Start-G-Code in prusaslicer könnte in etwa so aussehen:
> 
> ```
> M104 S0 ; set extruder temp to zero
> M140 S0 ; set bed temp to zero
> M190 S0 ; wait for bed temp to zero
> M109 S0 ; wait for extruder temp to zero
> 
> START_PRINT EXTRUDER={first_layer_temperature[initial_extruder]} BED=[first_layer_bed_temperature] FILAMENT=[filament_type]
> ```
> 
> Dies würde dazu führen, dass ein gcode beim Slicen so aussieht:
> 
> ```
> M107 ;turn fan off (i have not found a way to get rid of this but i can't see this do any damage)
> ;TYPE:Custom
> M104 S0 ; set extruder temp
> M140 S0 ; set bed temp
> M190 S0 ; wait for bed temp
> M109 S0 ; wait for extruder temp
> 
> START_PRINT EXTRUDER=215 BED=60 FILAMENT=PLA
> G21 ; set units to millimeters
> G90 ; use absolute coordinates
> M83 ; use relative distances for extrusion (this is toggleable in prusa slicer)
> ```
> 
> Dann könnte der Rest von Ihrem START_PRINT-Makro erledigt werden. Und stellen Sie sicher, dass G21 G90 und M83, das automatisch nach START_PRINT gedruckt wird, nichts für Sie ruiniert.

This doesn't work for me. My printer doesn't do anything with your GCode. As I wrote above, I've tried everything but nothing has helped. I've been sitting in front of the PC for several hours and tried it out. It can't be that difficult for programmers to take over the function of the Superslicer.

Biz810, 2022-09-27:
i found my problem. I had my start Gcode in the filament settings, the GCode doesn't work there. After I took over the start Gcode in the printer settings, it finally works now.

shame on me

Still, the feature would be nice.

mashuptwice, 2023-01-27:
> I don't understand what your FIRST_LINE_START will do, but if you want to suppress the M190 and M104 commands, use them in your custom gcode. So if you prepend your custom gcode with
> 
> ```
> M104 S0
> M190 S0
> ```
> 
> PrusaSlicer will then not set the temperature for you.
> 
> > the resulting behavior is unwanted and, not only unclear to users, confusing and frustrating
> 
> I don't think that setting temperature is unclear or confusing and I would assume slicer to do it for me.

This will not work, as the custom GCode from the printer tab will be inserted before the generated. One would need to do this in the filament settings for each single filament.

Chimorinxx, 2023-03-16:
@ghost
i have it setup the way you want to have it. i assume by bed leveling you mean bed mesh
```
M104 S120 ; set temporary nozzle temp
M140 S{BED_TEMP} ; set final bed temp
G28 ; home all axis
TEMPERATURE_WAIT sensor=heater_bed MINIMUM={BED_TEMP - 5} ; if bed temp is lower than specified, wait for heat-up`
G29 ;mesh
M104 S{EXTRUDER_TEMP} ; set final nozzle temp 
M109 S{EXTRUDER_TEMP} ; wait for nozzle temp to stabilize
G1 X5 Y10 Z0.2 F3000 ; get ready to prime
```
There are some lines missing that are not important for this reply.
i hope i could help you out. it didn't matter if i had them in the start gcode or in my macro

shawe, 2023-03-19:
I need to put this as start gcode:

```
; Solve PrusaSlicer BUG adding temperatures that it wants
M140 S45   ; Set Bed Temperature
M104 S160  ; Set Hotend Temperature
M190 S45   ; Wait for Bed Temperature
M109 S160  ; Wait for Hotend Temperature

START_PRINT EXTRUDER_TEMP={first_layer_temperature[0]} BED_TEMP={first_layer_bed_temperature[0]}
```

To prevent that PrusaSlicer put my bed to 90ºC before bed level when printing ABS.

My Klipper macro have a third parameter with BED_TEMP_PREV with default value set to 45, and four parameter EXTRUDER_TEMP_PREV to 160, it is used to pre-head bed and hotend to allow me a good level.

Because PrusaSlicer decides that I miss this values, it sets to what it wants, like disabling the fan 2 times, when only 1 is really needed

```
M107 ; disable fan
;TYPE:Custom
; Solve PrusaSlicer BUG adding temperatures that it wants
M140 S45   ; Set Bed Temperature
M104 S160  ; Set Hotend Temperature
M190 S45   ; Wait for Bed Temperature
M109 S160  ; Wait for Hotend Temperature

START_PRINT EXTRUDER_TEMP=245 BED_TEMP=90
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
; Filament gcode
M107 ; disable fan
;LAYER_CHANGE
```

If I don't put explicitly something like:

```
M140 S45   ; Set Bed Temperature
M104 S160  ; Set Hotend Temperature
M190 S45   ; Wait for Bed Temperature
M109 S160  ; Wait for Hotend Temperature
```

It adds auto-magically something like:

```
M140 S{first_layer_bed_temperature[0]}
M104 S{first_layer_temperature[0]}
M190 S{first_layer_bed_temperature[0]}
M109 S{first_layer_temperature[0]}
```

In this way, I need to put 2 times the same gcode, that I was controlling from a macro from my printer. If I replace the start_gcode, why you are considering that I'm missing something and added without any warning to confirm that unexpected behavior?

lukasmatena, 2023-04-02:
Setting temperatures automatically is optional since [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena