prusa3d/PrusaSlicer #2423: 3MF file stuck on mesh triangulation

jzkmath, 2019-05-30:
### Version
PrusaSlicer 2.0.0+win64

[image]

(I can add the rendering info if needed)

### Operating system type + version
Windows 10 Home, 64 Bit (full info below)

[image]

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S MMU2S 3.7.1, 1.0.5 
Slicer settings and profiles are more or less stock. Only changes made were to the Elephant's foot compensation value. Filament setting is Prusament PLA with the unloading speed changed to 120 and price to 23 (named eSun PLA+ V1)

### Behavior
When I tried to slice a .3MF file I loaded into PrusaSlicer, the program hung on "triangulating mesh". After waiting an hour with no sign of progress, I quit PrusaSlicer and restarted the program. After uploading the STLs into PrusaSlicer and reapplying the settings manually, the program sliced it in a matter of seconds.

Also, pressing cancel during the slicing would cause the program to stop responding and I would have to force the app to shutdown.

[image]

[image]

#### Project File (.3MF) where problem occurs
Here is a ZIP file with the 3MF file inside (as well as the two STLs used. They were split into multiple parts, and one was repaired using NetFabb within PrusaSlicer)

[GearCoin Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/3237722/GearCoin.Issue.zip)

Perhaps its this file specifically, but it still happened and might be a bigger issue. I would hate to find out that 3MF files put on PrusaPrinters don't work for others.

lukasmatena, 2019-05-31:
I managed to reproduce the issue. The problem is not the 3MF loading, it is really the elephant foot compensation. Turning it off or setting a bit lower value solves the issue. I agree that PrusaSlicer should do better than just get stuck on it, but I don't think you would get sensible results with this high compensation anyway. So as a quick solution, I would advise you to lower the compensation value. Does that work for you or is there a reason to really set it this high?

jzkmath, 2019-06-03:
> I managed to reproduce the issue. The problem is not the 3MF loading, it is really the elephant foot compensation. Turning it off or setting a bit lower value solves the issue. I agree that [...]

I adjusted the Elephant's Foot compensation slightly and the problem disappeared. 
As to why it was so high, I was in the process of tuning the value and honestly didn't think that 0.4 was a really large value. I wouldn't know as there is very little explained on the parameter or typical values used.

bubnikv, 2020-01-02:
The elephant foot compensation improvements are part of PrusaSlicer 2.2.0-alpha2. Closing.

closed 2020-01-02 by bubnikv