prusa3d/PrusaSlicer #2425: Rafts under purge towers

pearson, 2019-05-30:
### Version
2.0.0+win64

### Operating system type + version
Windows 10 build 17763

### 3D printer brand / version + firmware version (if known)
HICTOP 3DP17, Marlin 1.9+

### Behavior
If printing a raft and a purge tower is chosen, a raft with interface layers is created for the main object and for the purge tower. The interface layer weakens the base of the purge tower, and I've had a purge tower break free during printing at that interface layer.

Ideally there should be solid layers under the purge tower, rather than interface layers. There's no obvious reason to remove the raft from a purge tower after printing.

This may be a new feature request, or it may be a bug report, depending on one's point of view

#### Project File (.3MF) where problem occurs
Any project with a raft and a purge tower.

bubnikv, 2019-06-03:
I don't quite understand. A project file and screenshots would help.

AFAIK we print full base under the wipe tower.

rtyr, 2021-02-01:
The raft is not generated for the purge tower, first layer of the wipe tower is solid. Closing.

closed 2021-02-01 by rtyr