prusa3d/PrusaSlicer #2426: Unable to slice in command line mode: "Mixing configurations for FFF and SLA technologies"

bobstro, 2019-05-30:
### Version
PrusaSlicer 2.0.0+

### Operating system type + version
MacOS 10.14.5 Mojave on 27" iMac 5K

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 Mk3 on FW 3.7.1-226

### Behavior
Attempting to slice from command line with multiple .ini files that worked with Slic3rPE:

```
/Applications/Original\ Prusa\ Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer \
--printer-technology FFF \
--center 125,105 \
--export-gcode \
--loglevel 0 \
--info \
--load printers/printer_mk3.ini \
--load nozzles/0.20mm.ini \
--load filaments/PLA.ini \
--load settings/settings_common.ini \
--load settings/settings_012_layers.ini \
--output output/PLA/0.20mm/20mmcube.gcode \
models/normal/20mm_Cube.stl
```
Fails with message:

`Mixing configurations for FFF and SLA technologies`

No indication is given over what specific parameters in the ini files fail, and `--loglevel` or `--info` do not produce helpful output.

A simplified version works using a `config.ini` generated with PrusaSlicer:

```
/Applications/Original\ Prusa\ Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer \
--printer-technology FFF \
--center 125,105 \
--export-gcode \
--loglevel 0 \
--info \
--load ~/Desktop/config.ini \
--output output/PLA/0.20mm/20mmcube.gcode \
models/normal/20mm_Cube.stl
```
I have tried eliminating the smaller config files, but there are no indications of what parameters do not work with PrusaSlicer compared to Slic3rPE. There is no debug output, nor a `.tmp` file to review. Previous versions of Slic3rPE work well. I did have to change a few parameters for PrusaSlicer (center, printer-technology. There is no indication of what "Mixing configurations" means or what parameter is needed to avoid it.

#### Project File (.3MF) where problem occurs
Any STL source file produces the same error.

bubnikv, 2019-06-03:
please provide all the data (profiles, STLs etc), so we can reproduce the
issue. Also you may help by trying to minimize the set of parameters, at
which you still get this issue.

bobstro, 2019-06-05:
I've done some more testing. It's a problem if you specify more than one config file in the command line using `--load`. This works:

```
% /Applications/Original\ Prusa\ Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer \
--export-gcode \
--center 125,105 \
**--load config1.ini** \
--output 20mmcube.gcode \
models/normal/20mm_Cube.stl
10 => Processing triangulated mesh
20 => Generating perimeters
70 => Infilling layers
30 => Preparing infill
85 => Generating support material
88 => Generating skirt
90 => Exporting G-code to 20mmcube.gcode
Slicing result exported to 20mmcube.gcode
```
And this works:

```
% /Applications/Original\ Prusa\ Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer \
--export-gcode \
--center 125,105 \
### --load config2.ini \
--output 20mmcube.gcode \
models/normal/20mm_Cube.stl

10 => Processing triangulated mesh
20 => Generating perimeters
70 => Infilling layers
30 => Preparing infill
88 => Generating skirt
90 => Exporting G-code to 20mmcube.gcode
Slicing result exported to 20mmcube.gcode
```

But this fails with a cryptic message:

```
% /Applications/Original\ Prusa\ Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer \
--export-gcode \
--center 125,105 \
**--load config1.ini \
--load config2.ini \**
--output 20mmcube.gcode \
models/normal/20mm_Cube.stl

**Mixing configurations for FFF and SLA technologies**
```

Also, the `--no-gui` option seems to have removed, and no gui is launched if `--export-gcode` is specified. Is this behavior change documented anywhere?

bubnikv, 2019-07-29:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/3b21c64c2e0d30c8725a4100475008f1c62743e7
Closing

closed 2019-07-29 by bubnikv