prusa3d/PrusaSlicer #2427: Solid infill that hops over holes/gaps so that all lines are co-linear and only turn at the edge of a layer

jpswensen, 2019-05-30:
### Version
2.0.0+

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
I make a lot of flat parts with screw holes in them. The default rectilinear behavior is that the solid infill layers will make edge-to-edge passes when possible. However, when it runs into a gap/hole, instead of hopping over the hole, it goes back the other way. The result is that the line spacing is just a little bit different on paths that have a hole, versus those that don't. This can even be seen in the layer view of PrusaSlicer, not just in the printed part, so I think it is more than just the "mowing the lawn in different directions" effect.

An example of this can be seen [here](https://imgur.com/a/01p6WGK).

### Feature Request
So, my question is regarding how easy it would be to add an infill parameter to modify the behavior and have it hop over gaps/holes and continue along the same co-linear path until it reaches the outermost boundary. [Someone on Reddit](https://www.reddit.com/r/prusa3d/comments/buplj9/how_to_get_solid_infill_to_lift_up_and_skip_over/eph25pn/) pointed me in the right direction for how to add a new infill mode, but I think I need a little more insight into how the infill code works before I can start hacking on a new mode. Is there someone willing to give me a bit of mentoring about how the infill _generate functions work so that I can try to implement this myself?

Or, is there a really simple fix from editing the regular rectilinear infill?

bubnikv, 2019-06-03:
The pattern you are seeing at the first image is a shading issue, and it has nothing to do with the printing quality. We may look into improving shading of the print path preview, the shading artifacts are an effect of trying to share data between successive linear segments to conserve GPU memory.

The effect you are seeing on the print is an effect of pushing the excess of plastics in the direction of the newly extruded lines. If the direction changes, then the excess is pushed the other way around and an ugly ridge like artifact is created.

espr14, 2019-07-13:
The different look is caused by time shift between neighboring blocks of infill. The hops will remove the problem but it will bring oozing. Or you have to retract a lot, or you have to avoid holes. With big holes it will increase print time. Still, it is nice suggestion which makes sense to implement.

bubnikv, 2019-09-05:
> The pattern you are seeing at the first image is a shading issue, and it has nothing to do with the printing quality. We may look into improving shading of the print path preview, the shading artifacts are an effect of trying to share data between successive linear segments to conserve GPU memory.

> The effect you are seeing on the print is an effect of pushing the excess of plastics in the direction of the newly extruded lines. If the direction changes, then the excess is pushed the other way around and an ugly ridge like artifact is created.

The first issue should be fixed in the PrusaSlicer 2.1.0-rc.
The second issue is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2537

Closing this one.

closed 2019-09-05 by bubnikv