prusa3d/PrusaSlicer #2428: Copying support enforces/blockers

DeKemp, 2019-05-31:
### Version
2.0.0+win64

### Operating system type + version
Windows 10 pro 64bit

### 3D printer brand / version + firmware version (if known)
Not applicable, 

### Behavior
Problem: Copying support enforces/blockers after to be printed object hast been orientated (place surface on surface). will cause the enforcer/blocker to get an offset in position.

Normal behavior: result, coping works as expected
[image]

Abnormal behavior:
Step 1 change orientation of object
[image]

Step 2, add support enforcer/blocker
[image]

Step3, copy said enforcer/blocker
[image]

Result. Copied encorcer/blocker will not stay in the same position x position.

bubnikv, 2019-06-03:
The copied modifier meshes are shifted on purpose, so they will not overlap
with the original. The behavior is similar to 2d painting software as MS
Paint Brush.

bubnikv, 2019-08-09:
Fixed by @enricoturri1966:
    1) Reworked logic for pasting volumes
    2) Fixed paste of volumes into different objects
    3) Do not apply offset when pasting into the copied object
    4) Keep source transformation matrix and relative positions when copy/pasting volumes into another object

The fix will be part of the upcoming PrusaSlicer 2.1.0-alpha1. Closing for now. Please feel free to reopen if you find any issues / inconsistencies with the current fix.

closed 2019-08-09 by bubnikv