prusa3d/PrusaSlicer #2430: Interest in installer exe?

rejas, 2019-05-31:
Hi there
I created an installable exe for Pruser since some friends find it useful instead of unzipping somewhere. 
 So I was wondering if you are interested in using the nsis-installer-scripts to build a exe file for your software automatically? In case you want to try them out you can find them on my private homepage: https://veeck.de/personal_software.html#nsis

vojtechkral, 2019-06-01:
PrusaSlicer is part of an installation package that you can download on [PrusaSlicer homepage](https://www.prusa3d.com/prusaslicer#intro). Is that for some reason not suitable for you?

rejas, 2019-06-03:
oh, never noticed that, was only looking at your github pages the whole time :-) thx for that and all the great work you are doing!

closed 2019-06-03 by rejas