prusa3d/PrusaSlicer #2432: crashes

jasonp568, 2019-06-01:
### Version
2.0.0+win64

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
mk3 with newest firmware 

### Behavior
crashes when slicing 
load up model and attempt slicing
[_Vive_magnetic_controller_dock_-_No_Logo.zip](https://github.com/prusa3d/PrusaSlicer/files/3243471/_Vive_magnetic_controller_dock_-_No_Logo.zip)

enricoturri1966, 2019-06-02:
Running PrusaSlicer in debug mode I got an assert at line 105 of double Flow::mm3_per_mm() const (found negative res)

These are the values found:
res = -0.58680156250385607
this->width = 0.419999987
this->height = 2.90000010
this->bridge = false

jasonp568, 2019-06-02:
And that is why my slicer crashes? What does it mean? My slimpfy 3d has no problem

lukasmatena, 2019-06-03:
Yes, that is why your PrusaSlicer crashes. For some reason, it thinks that first layer is 2.9mm high, which is impossible and the skirt generating algorithm cannot cope with that. Turning off the skirt bypasses the crash, but the resulting object would be printed in mid-air.

The "some reason" is most likely the fact that your model is badly broken. Cutting it by the cut tool at 0 height made the issue disappear for me (but other issues may still  be present). I suggest you fix your model by Netfabb (you're running windows 10, so it is easy), this will almost certainly solve it.

We should improve the model checking/fixing routines in PrusaSlicer one day.

bubnikv, 2019-09-09:
Current PrusaSlicer 2.1.0-rc processes this model correctly. 
Also PrusaSlicer 2.1.0 has been extended to not crash, but detect the cases where the 1st layer produced an empty slice.

Closing.

closed 2019-09-09 by bubnikv