prusa3d/PrusaSlicer #2433: Print time estimation increases with increased travel speed

Rudgas, 2019-06-01:
### Version
2.0.0

### Operating system type + version
Linux 5.1.3-arch2-1-ARCH Arch Linux

### 3D printer brand / version + firmware version (if known)
Hypercube Evolution, Marlin bugfix (current)

### Behavior
Print time estimation increases with increased travel speed.
For example the xyz calibration cube takes about 36 minutes at 150mm/s travel speed and 1h10 at 1500mm/s travel speed. At 30mm/s travel speed it goes down to 21 minutes.

Using the estimation from https://www.gcodeanalyser.com actually shows a reduction in printing speed when increasing travel speed. Although it is minor as accel settings are limiting here.

Sebastianv650, 2019-06-01:
Strange, looks as #1738 is back. This issue was present in the "old" Slic3r PE and gone in the PrusaSlicer beta. Now in final, I can reproduce it one more.

bubnikv, 2019-06-03:
Please attach your project files. It may be, that the cooling slow down
kicks in, which skews the time estimation results.

Rudgas, 2019-06-03:
There you go (Had to change the ending from 3mf to txt)

[xyzCalibration_cube.txt](https://github.com/prusa3d/PrusaSlicer/files/3246886/xyzCalibration_cube.txt)

bubnikv, 2019-06-03:
The slow down will certainly kick in. You may try to disable the print slow down at the filament profile.

The issue here is, that the print slow down is not aware of the firmware acceleration profiles, therefore if you increase travel speed, the print slow down kicks in in slicer, which in turn will increase the print time.

Rudgas, 2019-06-03:
I don't quite get the what you are saying, but in any case auto-cooling is disabled in my filament profile.

bubnikv, 2019-06-04:
If the auto cooling is disabled, then my previous comment does not apply.

Anyways, you have the Machine Limits - Maximum Feedrates set low, they limit the travel feed rate. Giving a travel of 1500mm/s has no effect, if the maximum feedrate in X and Y axes is set to 150mm/s in Machine Limits.

Rudgas, 2019-06-05:
That's what I was thinking too, machine limits should take precedence, but I wouldn't expect a decrease in print time going above 150mm/s anyways, especially not for the xyz calibration cube.

enricoturri1966, 2019-06-06:
Working on it -> https://dev.prusa3d.com/browse/SPE-958

bubnikv, 2019-08-07:
This issue will be fixed by the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-07 by bubnikv