prusa3d/PrusaSlicer #2437: Spurious error message when saving to GVFS-mounted DAV server

stuartpb, 2019-06-02:
### Version

2.0.0

### Operating system type + version

GNOME under Arch Linux

### Behavior

I have a FlashAir that I save my print files to mounted using GNOME's Files browser via [GVfs](https://en.wikipedia.org/wiki/GVfs) (so it's mounted to `/run/user/1000/gvfs/dav:host=192.168.0.33,ssl=false`).

When I hit "Export G-code" and choose the FlashAir as the destination to export to, everything proceeds fine and the file is saved to the card successfully, but an error message still pops up afterward saying "Copying of the temporary G-code to the output G-code failed".

rtyr, 2021-02-01:
Is this issue still valid?

stuartpb, 2021-02-10:
This may still happen, but it's no longer applicable to my use case, as I no longer save files through GVFS since PrusaSlicer got direct FlashAir support.

bubnikv, 2021-03-12:
We are now seeing a similar issue when running PrusaSlicer on ChromeOS / crostini Linux container and exporting to removable media.

bubnikv, 2021-03-16:
We cannot be sure, but hopefully the issue was setting the file attributes after the file was copied. This issue is similar to what we experienced on Chromebooks / Crostini Linux virtualization:

01406fd52182d72cf1789174dd64ee4284e9f2cc

    Worked around some quirky Linux file system issues. Namely
    the Chromebooks share their file system to Linux using the 9p file
    system, which does not support setting file ownership. Newly PrusaSlicer
    will detect platform and it will not panic if copy_file() cannot set
    file ownership after copying. It just logs the incident, and on
    chromebooks the loglevel for that incident is "Info", not "Error".

Closing this issue. If you still experience this issue with the upcoming PrusaSlicer 2.4.0, please let us know.

closed 2021-03-16 by bubnikv