prusa3d/PrusaSlicer #2438: PrusaSlicer Crashes upon oppening with "'=' character not found in line error"

Eashley, 2019-06-02:
### Version
_2.0.0_

### Operating system type + version
_Windows 10 Pro 1803_

### 3D printer brand / version + firmware version (if known)
_MK3K 3.7.1 _

### Behavior
* After Downloading the new Prusaslicer app on version 2.0, The Application does not start. Instead it throws up an error via 2 dialog boxes:

1. Unhandled unknown exception; terminating the application.
[image]

2. internal error: <unspecified file>(22): '=' character not found in line.
[image]

* My steps were as follows:
1. Install prusaslicer via the https://www.prusa3d.com/drivers/ page using the 2.2.6 Drivers and Apps installer.
2. After completing, merely starting the app causes this issue.

Notes: 
* I also attempted using the zip file located on https://github.com/prusa3d/PrusaSlicer/releases. 2.0.0 and 2.0.0-rc produce the same result; however, Slic3r Prusa Edition 1.42.0-beta2 works as expected.
* I attempted to try both the 32bit and 64bit versions. Both produce the same result.
* Using the command like command line executable ( prusa-slicer-console ) returns the same error message: 
[image]
* I noticed that this seems similar to the error found in https://github.com/slic3r/Slic3r/issues/4724 
** I looked into HOME\Prusa3D\PrusaSlicer\resources\profiles\PrusaResearch.ini had any lines that contained no "=" sign; however, after a quick look I couldn't find one. Attached is my file if that helps.
[PrusaResearch.txt](https://github.com/prusa3d/PrusaSlicer/files/3244996/PrusaResearch.txt)

If I missed any other details, or any further research needs to be done on my end, let me know. I could not find any other error logging in the program directory.

Eashley, 2019-06-03:
Looks like something got corrupted in my AppData directory. Deleting the prusaslicer directory there seems to have done the trick. I am still unsure as to the "why", but if someone else stumbles upon this, try deleting C:\Users\{UserName}\AppData\PrusaSlicer*.

ansi9091, 2019-08-13:
I have the same error after many downloads and re-installations. Any advice? I deleted "AppData\PrusaSlicer*" but is doesnt help.

vojtechkral, 2019-08-14:
Guys, deleting `%AppData%\PrusaSlicer*` doesn't help use figure out what was wrong or how it got corrupted in the first place. Please always move/rename the directory instead of deleting it so that later you could shared the corrupted dat with us.

@ansi9091 If you still see the same error, could you perhaps zip up your `%AppData%\PrusaSlicer*` and share it with us?

paynterf, 2019-08-20:
Same problem, but I copied rather than deleting the problem folder.  I have attached a zip file.  I noted that the .ini file seemed corrupted and filled with NULL chars rather than human readable text as expected.

Hope this helps,

Frank
[CopyOfPrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/3521192/CopyOfPrusaSlicer.zip)

bubnikv, 2019-08-21:
@paynterf Thanks for heads up. It is mysterious to me how that could have happened.

The PrusaSlicer.ini is first written into a unique file with a suffix given by the PrusaSlicer process ID. After it is successfully written, it is renamed to PrusaSlicer.ini using some Microsoft API. It looks like the Microsoft API produces these broken files. 

I have tried to improve the situation with fd3fe75d1ca4ddf1a36eb89d7210a4ec0e8f4afa :

    Reworked the rename_file() function on Windows to work reliably and
    atomically. The code was taken from the llvm project, it is complex
    and hopefully it covers all the Windows file system quirks. Vojtech
    has highest hopes, that this will fix the various PrusaSlicer.ini
    file corruptions.

bubnikv, 2019-08-24:
While we have no explanation on how the PrusaSlicer.ini file could have gotten corrupted, we suspect that the Windows API for renaming files is not atomic and it may in some cases lead to file corruption. We replaced the "file rename" function with a complex function from the llvm project with fd3fe75. Let's hope it will fix this mysterous issue.
Closing, the fix will be part of the PrusaSlicer 2.1.0-beta.

closed 2019-08-24 by bubnikv

ErasmusJuve, 2020-01-15:
I had the same error, I solved it renaming the PrusaSlicer.ini file with COPYPrusaSlicer.ini, opened PrusaSlicer and asked me to configure the printer again (because it loses the initial configuration), I closed the application, opened both .ini files and copied the sections of [presets] and [vendor: PrusaResearch], I went back to open PrusaSlicer and everything worked perfect

I found that a worthless field, "last_output_path =" this could have been the error in my case, I don't know about you but try this

bubnikv, 2021-07-02:
Transactional saving of PrusaSlicer.ini implemented in PrusaSlicer 2.3.2-beta.