prusa3d/PrusaSlicer #2440: Dual Extruder

biskero, 2019-06-02:
Hello,

if I set dual extruder there is no option to do the following:
1 - cannot set different material for each extruder in Filament Type
2 - cannot set density and cost per filament type,
3 - no stats after slicing per filament

bubnikv, 2019-06-03:
For a dual extruder, you should uncheck the "single extruder multi-material" checkbox.
Other than that, I don't understand your issue. All should work. Would you please attach your project file?

biskero, 2019-06-03:
I do have unchecked the multi material.
The issue is that in Filament Type I cannot select 2 different materials for the 2 different extruders. I can select only one, but I have 2 different materials.
If I have 2 different materials, for example HIPS for support and CF, it would be good to have separate stats like density and cost per each materials, so that at the end of slicing there are information for the 2 different materials. This way I would really know the weight and cost of the print for the actual print and for the support material.

YuSanka, 2019-06-03:
Thanks for your report. There is a small bug (the "gear" button near the filament preset has no expected effect in some cases). Bug will be fixed in the upcoming patch release. But everything else works as expected. 
Understand that "Filament Settings" tab is used just for editing the preset, not for its selection. 
So, to solve your problem, you should select wanted preset for each filament in the right panel in Plater. And on the "Filament Settings" tab just edit and save options (density and cost per filament type) for each of the selected presets separately.

YuSanka, 2019-06-03:
Thanks for your suggestion. We'll think about it. But this issue is resolved? Can it be closed?

biskero, 2019-06-03:
yes of course.
Hope that would be implemented

closed 2019-06-03 by biskero