prusa3d/PrusaSlicer #2441: what can i do to never see this look?

akust, 2019-06-02:
### Version
2.0.0+ 201905201652

### Operating system type + version
macOS Sierra

### 3D printer brand / version + firmware version (if known)
(it's about mac software only)

### Behavior
* _Describe the problem_
romdomly but definitly too often Slic3r/PrusaSlicer opens in super oldschool default printbed - but I _never_ had something else than a Prusa printer NOR did I ever setup any other Printer in the Software. And still the software insists to randomly change the view too the gray no-name view. It's confusing and I need to outwit the Software by extra changing templates, close and restarting the app. 
I'd like to ask for
- Is there a text based option to correct something in the .ini files? Please give me a tip what I maybe could correct myself. (as you can see, this look drives me nuts and I'm ready to edit the ini files if needed)
- add a batch process "re-make all templates orig prusa" or just never even consider something else than a prusa because it's all about prusa and there is no other printer in my list/ownership
- avoid accidential overriding settings resulting in the ugly grayish bed view

* _Steps needed to reproduce the problem_.
I don't know, sorry it's freaqing random. I can offer the last project before it showed the wrong bed on restart (see zip below)

(..no command line where used, just standard mac app)

* _Expected Results_
Slicer shows the elegant Prusa print bed
* _Actual Results_
Slicer show ugly grayish pixel testpicture as printbed replacement 

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
ok! this is what the actual PrusaSlicer looks like:
[image]

_Is this a new feature request?_
maybe: please add a function: override ALL Custom Templates to show the origina Prusa bed ALWAYS/ delete the test picture and/or make it never show up again

#### Project File (.3MF) where problem occurs
after slicing and saving following project I closed the App. Now it opened again showing the ugly bed.

[Sp4ceRings-NI_JH+LF_PLA_.25mm_PLA.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3245350/Sp4ceRings-NI_JH%2BLF_PLA_.25mm_PLA.3mf.zip)

and btw, regarding the following hint - uploading .3MF is not supported. You should add "zip before rying upload" or something..
"_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_:

bubnikv, 2019-06-03:
You may be low on GPU memory. You may try 
https://github.com/prusa3d/PrusaSlicer/issues/2299#issuecomment-494493816

akust, 2019-06-03:
thank you for the tip. But no, it's not related to GPU.
I think/hope I found the problem myself now: 
in .ini files look for "inherits = " (often in line 17)
If rest of the line is empty Slicer assumes a foreign printer(why?) and switches to the gray default bed.
I assume that sometimes saving a template just leaves it empty resulting in "you're not prusa" look.

Add "Original Prusa i3 MK3" (without "") to the line and (after reloading app once more) the template looks correct.
I now went through all my .ini files and corrected it (really often!) - hope to see that bad bed never again.

And I ask here just to double check why this content get's lost at all and ensure that it stays there, or even better: **if empty (why ever) - still show the Prusa Bed** - why not??

apart from that and other little issues: I like Slicer very much! Thanks for your efforts!

bubnikv, 2019-08-01:
I don't know how the "inherits" field got lost in your case, it should not. It may have possible happened, if we had renamed the system printer profiles. We should strive to not to.

bubnikv, 2020-03-20:
This issue is likely due to misconfiguration, which may have happened during PrusaSlicer upgrade. I suppose we can close this issue.

closed 2020-03-20 by bubnikv

3Dietrich, 2020-03-20:
well.. this was 2.0.0, i think a version around the release.. it disappeared/the Prusa bed keeps visible since then.. bug not existing anymore
I should have closed this myself..