prusa3d/PrusaSlicer #2443: Crash when slicing object with support

Raniz85, 2019-06-03:
### Version
2.0.0+win64

### Operating system type + version
Windows 10 1803

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
When slicing with supports PrusaSlicer crashes with the progressbar at around 90%.

I can see the sliced model show up but then PrusaSlicer crashes without any error messages.

#### Project File (.3MF) where problem occurs
I included the STL as well
[Angled Hook.zip](https://github.com/prusa3d/PrusaSlicer/files/3247000/Angled.Hook.zip)

lukasmatena, 2019-06-03:
I tried in debug mode and it trips on the same assert as yesterday's issue #2432. We tried fixing the model through netfabb but it failed to produce any result. Are you sure the model is valid?

Raniz85, 2019-06-03:
Sure? No, I don't know how to be that. But I've never had any issues exporting models from Fusion360 before and I just printed the model using Simplify3D.

Raniz85, 2019-06-03:
I can't seem to slice anything that includes supports now. I've tried four different models - two of which I've successfully sliced with the same version of PrusaSlicer before in the same orientation that I tried now, and one in a different orientation and all of them crashes the program.

Raniz85, 2019-06-03:
I've found the issue. I had wrongfully set my printers max layer height to 3.2 mm instead of 0.32 mm. After correcting that everything, including the original model, slices correctly with supports.

closed 2019-06-03 by Raniz85