prusa3d/PrusaSlicer #2445: Unable to save project

ochm, 2019-06-03:
### Version
2.0.0+win64, 1.42.0-beta2+win64-full

### Operating system type + version
W8.1 64bit CZE

### 3D printer brand / version + firmware version (if known)
i3 MK3 3.7.1

### Behavior
Attached .stl can be resized (optionaly), can be sliced, can be gcode exported,
but the project with this .stl can't be saved to 3mf.
Every try (in both versions) leads to unresponsive application and CPU (i5-3750 3.4GHz) utilisation at 30%, for long time, (tens of minutes) without end. Application must be killed.
Any other .stl goes OK.

#### Project File (.3MF) where problem occurs
[PenHolder.zip](https://github.com/prusa3d/PrusaSlicer/files/3248841/PenHolder.zip)

bubnikv, 2019-06-04:
I see an endless loop due to a messed up triangle mesh structure. It is
certainly not good, but I suppose it is a very rare case. I am
investigating it further.

bubnikv, 2019-06-04:
It would not happen, if the input model was consistent.

bubnikv, 2019-06-04:
Thanks for the data. I am testing a fix 3ab886b747c4742d1662a415d683857003c0388b

The problem is in the admesh library, which Slic3r (and PrusaSlicer) use to load and fix a mesh. Your mesh is tiny, but contains a large number of triangles. When slicer tries to close holes and fix the mesh, it degenerates some triangles into zero area, and this case was not correctly handled by the admesh library.  While I believe I patched some holes, I bet there are some holes left to be patched. The admesh library is quite old and newer libraries use a more robust approach, we shall likely get rid of the admesh library one day.

ochm, 2019-06-05:
Yes, with Netfabb repaired .stl problem disappeared.

rtyr, 2021-02-01:
Fixed. Closing.

closed 2021-02-01 by rtyr