prusa3d/PrusaSlicer #2446: crash when exporting configuration bundle

jbernardis, 2019-06-03:
### Version
2.0.0+win64 - also same behavior on ubuntu allimage

### Operating system type + version
win 10 64 bit, ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
custom printer running marlin - not printer related

### Behavior
when exporting configuration bundle, bundle is exported fine, but then the program crashes.  Every time on both operating systems

#### Project File (.3MF) where problem occurs
not model specific

lukasmatena, 2019-06-04:
This might be a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2307. Have a look at that one and try if the suggested workaround works for you. If it does, you can close this issue because the bug has been fixed already.

jbernardis, 2019-06-04:
I'm not sure I follow.  I'm to define a bogus SLA  printer, and this is considered the fix?

Or are you saying that defining an SLA  printer will show that it is the same issue, and that a real fix is coming.?

JohnOCFII, 2019-06-04:
> Or are you saying that defining an SLA printer will show that it is the same issue, and that a real fix is coming.?

That process is a workaround until a fix can be tested and released.

jbernardis, 2019-06-04:
Nope

I added an SLA printer to the available printers and tried to export the
bundle and it crashed.  I reloaded the program and the SLA printer was no
longer in the configuration.  So I re-entered it and exited the program.
This time when I reloaded, the configuration was correct, but when I tried
to export the bundle again, it crashed.  This same thing happened on both
Linux (Ubuntu 18.04LTS) and Windows 10.  I am using the 64 bit program on
each

jwvaughn, 2019-06-05:
I have the same as jbernardis including the suggested fix not working on two Win 64 bit Pro machines.

In addition, 2.0.0 will no import a confi bundle exported from the last release version giving the error "<unspecified file >(1469): '=' character not found in line"

Sembazuru, 2019-06-27:
I've found that when I save a configuration bundle and Plicer crashes (happens every time on two different Win10 laptops) the last 3 lines are:

```
[presets]
print = 0.20mm QUALITY MK3 (CDE)
sla_prin`
```

If I then delete these last 3 lines then I can import the bundle. It seems that Plicer is crashing when trying to save out my currently selected profiles for the SLA printer that I don't have (or have configured). I just checked that other bug report (2307) and it seems to be the same issue. So my temporary 'fix' of hand editing the config bundle file to delete the malformed presets section is another way to get around this.

This is very important for me because I use Plicer on two different laptops to create models for my one printer. One is down in the basement next to the printer and the other is upstairs next to my couch. It is very important for me to keep both computers synchronized to the same configuration bundle.

wildblue, 2019-07-13:
open the file  PrusaSlicer.ini in your config-folder
maybe also check the file  slic3r.ini

change
<
sla_material = 
sla_print = 
>
into
<
sla_material = Prusa Orange Tough 0.05
sla_print = 0.05 Normal
>
open prusa slicer and export your config bundle.

background:
the file vendor/PrusaResearcch.ini is invalid
following must be changed to work on initializing
<
[sla_print:*common*]
compatible_printers_condition = printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and
>
into
<
[sla_print:*common*]
printer_technology = SLA
compatible_printers_condition = 
>

and
<
[printer:Original Prusa SL1]
printer_technology = SLA
...
>
into
<
[printer:Original Prusa SL1]
inherits = *common*
>
this change results in the correct behaviour, but only if not run once without this fix.
the old config generates the empty config-valurs and these can only be fixed manually at the moment.

hey prusa-team, what is taking so long explaining this or releasing a bugfix? almost2 month since the last release.

don't you have enough developers? i would work for your company, having 10 years experience in c++, c#, digital image recognition, electronics, mechanics. have built my own 3d printer not long ago completely from scratch.

maybe i hear from you, i am located in germany, but could move when nessesary.

with the best regards in this short time

ben

lukasmatena, 2019-07-13:
> hey prusa-team, what is taking so long explaining this

The explanation is quite obvious: it is a bug. See also https://github.com/prusa3d/PrusaSlicer/issues/2307#issuecomment-494518628 for more detailed info. The next version will be released when it is finished.

>  i would work for your company

Thanks for your interest. I suggest going the usual way and contact our HR department directly through a web form at the bottom of [this](https://www.prusa3d.cz/volna-mista/) site and provide all relevant info.

bubnikv, 2019-08-08:
This bug will be fixed by the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-08 by bubnikv