prusa3d/PrusaSlicer #2448: Problem with g-code on i3MK2 MMU1

Laborky, 2019-06-04:
### Version

Prusa Slicer 2.0.0+win64

### Operating system type + version

Windows 10

### 3D printer brand / version + firmware version (if known)
i3 MK2MMU1; latest firmware

### Behavior

If we try to print the sliced G-CODE with soluble interface or full soluble supports, the printer shows error - incomplete file. It occurs with all models and if we tried to print it, there are errors like missiíng layers.

#### Project File (.3MF) where problem occurs

[Podzemi.zip](https://github.com/prusa3d/PrusaSlicer/files/3253185/Podzemi.zip)

rtyr, 2019-06-06:
Recently released firmware v3.2.3 should fix the issue with incomplete file check.

You can download it here https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.2.3

rtyr, 2020-05-21:
Closing, fixed with FW 3.2.3.

closed 2020-05-21 by rtyr