prusa3d/PrusaSlicer #2449: [Feature Request] Smart Layer Height based on horizonal surfaces

Mithrandil, 2019-06-04:
It's quite annoying to be forced to design objects with the flat surfaces height based on the layer height at which they will be printed (Specially now that the layer height table has been removed)

(e.g. slicing a 10,1 mm high cube with 0,20 mm layer height leads to a 10.2 mm high cube)

So, i suggest a smart modification of the layer height based on the distance of flat surfaces from the heated bed, exploiting the layer editing feature to make it automatic, 

(for instance, slicing a 10,1 mm high cube at "smart" 0.20 mm layer height could lead to 41 layers 0.20 mm thick + 10 layers 0.19 mm thick --> 10.1 mm high cube)

[image]

YuSanka, 2019-06-05:
Thanks for your suggestion. Actually, we working on rewriting of the layers table right now. And new implementation will be a part of one of next releases.

bubnikv, 2019-06-05:
I know the upstream slic3r has some automagic logic such as you have
described. The problem is, that it is only well applicable to simple
objects. For objects with multiple horizontal steps the algorithm will
likely be more involved and / or will not produce desirable results.

Mithrandil, 2019-06-05:
Well, even if there are multiple steps, a simple algorithm could just divide the object (i.e. create a row in the layer table) for each interval between two steps, and calculate the needed layer height accordingly, whithin set limits.

If the geometry is to complex to be solved autometically the algorithm could just give up with an alert, encouraging the user to check the layer height table by hand.

Mithrandil, 2019-08-23:
Here it is!

Python3 script which optimizes the layerheight in order to match right z values on flat surfaces (as per 1st post), approximated at 0.01 mm
[layer height only supported in in PrusaSlicer2.1.0-alpha1 for the moment]

usage: 
`python3 LH-optimizer.py <default_layer_height_in_mm> <file.stl>`
example:
`python3 LH-optimizer.py 0.40 mystlfule.stl`

output:
```
 0.0  -->  0.2  -  0.2
 0.2  -->  0.9  -  0.35
 0.9  -->  3.0  -  0.42
 3.0  -->  5.0  -  0.4
 5.0  -->  11.24  -  0.39
```

[LH-optimizer.py.txt](https://github.com/prusa3d/PrusaSlicer/files/3534916/LH-optimizer.py.txt)

Mithrandil, 2020-07-16:
New version of the script, with some improvements:

- Works directly with .3mf files instead of .stl (models can be oriented and scaled before optimization)

- Saves the layers directly in a new .3mf file named OPTLH_old.file_name.3mf
 
- Optional Parameter: max_layer_height (usually you want to set it at 0.75*nozzle-width)

`Usage:  NEW-LH-optimizer.py <file.3mf> <default_layer_height_in_mm> [max_layer_height_in_mm]`

[NEW-LH-optimizer.zip](https://github.com/prusa3d/PrusaSlicer/files/4933719/NEW-LH-optimizer.zip)

closed 2026-09-29 by prusaslicer-bot[bot]