prusa3d/PrusaSlicer #2450: PrusaSlicer 2.0 Sequential printing vs rotate object

angelshtilianov, 2019-06-04:
PrusaSlicer
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17134), 64-bit edition

3D printer brand: Prusa MK3/MK2.5 fw ver: 3.7.1

Behavior:
In sequential printing there are some discrepancies in objects distance calculation.
If I import the objects and set some specific clearance between the extruder and an object everything is fine, But if I import a rectangular object (sized X:69 Y:15.2 Z:24.6mm in my case) and rotate is 90 degrees around Z to get it printed with its short side by X (it should become X:15.2 Y:69 and Z:24.6). It visually does, but sequential printing seems is not aware about the rotation at all and generates warning if the objects are closer than their original X size (X size before rotation) + the clearance setting in sequential printing.
Easy way to reproduce: Import a rectangular object (just a box) with the dimensions mentioned above, rotate it 90 degrees around Z, add one more copy using the "+" button on the top menu (just in right of the paste button) and play with the objects. Now you can overlap them by Y, but you can not put them closer than 69 mm each to other on X, even the extruder clearance is set to zero.

Please find the file attached:
[holder.zip](https://github.com/prusa3d/PrusaSlicer/files/3254578/holder.zip)

bubnikv, 2020-02-06:
Do you have the issue with PrusaSlicer 2.1.1?
Do you have the issue with PrusaSlicer 2.2.0-alpha2?

closed 2020-02-07 by bubnikv

bubnikv, 2020-02-07:
The validation algorithm did not take object rotation into account by mistake. Fixed with 348bf390124a74dc5e86adcd99cc2f96f9ee1f11
The fix will be part of the PrusaSlicer 2.2.0-alpha4.
Closing.

commit b8c898bf (2020-02-07): Fixed a bug in validation of the FDM Print to check for extruder
collisions against already printed objects: Rotation of an object
was not being taken into account.
Fixes #2450 PrusaSlicer 2.0 Sequential printing vs rotate object
Fixes #3585 Always Sequential Printing Error (Regardless of setting, placement or object size)

Fixed some collisions in sequential print just after arrangement by
making the validation using a slightly lower extruder radius value
than the arrangement.

Refactored PrintObject coordinate system so that the PrintObject's
coordinate system is always centered in respect t [... cut]