prusa3d/PrusaSlicer #2453: Size/Move buttons are not displayed

TomaxMe, 2019-06-05:
### Version
PrusaSlicer
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17763), 64-bit edition

OpenGL installation
GL version:   1.1.0
Vendor:       Microsoft Corporation
Renderer:     GDI Generic
GLSL version: N/A
Installed extensions:
GL_EXT_bgra
GL_EXT_paletted_texture
GL_WIN_swap_hint

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17763), 64-bit edition

### 3D printer brand / version + firmware version (if known)
PrusaSlicer
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

### Behavior
First installation of the program. Size / Move etc buttons are not displayed. 
Attached pic.
[image]

bubnikv, 2019-06-05:
> OpenGL installation GL version: 1.1.0

Your graphics driver does not support OpenGL 2.0 or newer, and the automatic switching to the software rasterizer did not work. You may either start the application with a --sw-renderer parameter, or you may copy the opengl.dll from mesa directory next to prusa-slicer.exe. Please  understand that the software reasterizer will run much slower than the real hardware opengl rasterizer, therefore it is advisable to update your drivers.

We shall improve slicer to give at least some hints on this issue.

TomaxMe, 2019-06-05:
Thanks to point this out... That's happened in my multiple monitor setup via Dell 4 monitors hub. I've just disconnected this hub and restart the app then I can have all these buttons displayed. Then, I reconnected the Dell multi-monitor hub and I can see corectly also on my multi monitor setup.

rtyr, 2021-02-01:
Looks like this issue was resolved. Closing.

closed 2021-02-01 by rtyr