prusa3d/PrusaSlicer #2455: [Feature Request] Skirt/Brim for each individual object

th33xitus, 2019-06-05:
### Version
2.0.0 final

### Operating system type + version
Win7 x64

### Is this a new feature request?
Yes 

### Request
I think it would be benefitial for each object to have its own skirt/brim instead of wrapping one big single skirt/brim around all of the objects.

At the moment you get something like this when placing (in my case 3) objects on the plater:
[image]

The yellow marked area is completely useless, having no purpose and simply wasting material and time:
[image]

It would be smarter to wrap a skirt/brim around each individual object so the skirt/brim actually does what it should, either prime the nozzle or help with print-bed-adhesion.

### Additional Request
Maybe it is possible to implement a checkbox to get a skirt/brim in the inside of objects?
In case of my cubes with the hole in the middle the option to toggle a skirt/brim inside of the hole?

reinhapa, 2019-06-15:
I would suggest to add the skirt/brim possibility as a additional individual property. This way it would be possible to print multiple objects and only have a few of them having a skirt/brim.

th33xitus, 2019-06-15:
Would be a legit solution as well.
Implementing the Skrit/Brim options in the object settings so you can simply change it with a right click.

ghoss, 2019-07-02:
I second @reinhapa 's suggestion. This would solve the real nuisance of having to sort out and arrange collections of objects in two passes (one for all objects without a brim, and a second run for those which need one).

salmander, 2019-07-03:
Definitely 👍 from me. This would come in really handy, especially when you are adding multiple items on your build plate and some of them require brim to succeed and you don't want to waste material/time adding a brim to all of them.

watzi3, 2019-07-14:
I can only agree with @reinhapa. 
Being able to select a brim for each object on the plate would be a huge improvement for prints with different models.

bubnikv, 2019-08-01:
I believe this is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/286.
Currently PrusaSlicer calculates skirt and brim for the complete 1st layer, not per object.

ghoss, 2019-08-01:
@bubnikv Yes, it seems to be the same feature request.

th33xitus, 2019-08-01:
Yup it is.
Sorry for that but i wasn't aware that there was already such a request from 2 years ago ^^
I hope we will see this feature maybe in future updates?

mgx0, 2019-12-13:
bump

please add skirt around individual objects, like Simplify3D has. This skirt (when has more loops, like 10 or 20 with distance 0.3mm from the object) could help object adhesion. I have bought Simplify3D a year ago but I'm ready to migrate to Prusaslicer. This feature is really useful when trying to keep the objects on the plate - bigger surface

n8maher, 2020-01-15:
Hi everyone, I think this is a similar issue - PrusaSlicer won't add brim to a part nested inside another part.  I have attached a picture showing what I mean.

[image]

ThePirat91, 2020-03-08:
Hello, any news on this topic?
Brim for selective objects would be a really useful thing....
Thanks

rtyr, 2020-03-23:
I believe this is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/286. Closing.

wnovinger, 2020-08-30:
I agree with the suggestions above, but would suggest the scope be widened to include per-object rafts as well. Many of the projects I work on have small or otherwise not very stable objects which greatly benefit from rafts. Being able to print an entire assembly at once would be very useful. Thanks for listening and keeping this issue alive, **rtyr**!

22chrs, 2021-02-10:
Yes! Brim for inside objects (also if it is only one part with inside geometry touching the bulid plate) would be nice! :)

lukasmatena, 2021-08-12:
Upcoming PrusaSlicer 2.4.0-alpha1 will allow to set brim for each object separately. I don't think we will implement skirt per object in near future (if at all), I'm not sure if it even makes sense. Can I close the issue?

th33xitus, 2021-08-12:
@lukasmatena imho definitely individual brim > individual skirt ... i wrote skirt/bring because technically a brim is a skirt with 0 distance to the object. but i don't see any heavy usecase right now for individual skirts. maybe only for priming the nozzle between the parts if there is some oozing happening during travel moves.

wnovinger, 2021-08-12:
Glad to hear about decision re brims on per-part basis. I earlier suggested similar capability for rafts, too. Is there any consideration for this? TIA

rtyr, 2021-08-12:
> Glad to hear about decision re brims on per-part basis. I earlier suggested similar capability for rafts, too. Is there any consideration for this? TIA

It is already possible.

[image]

wnovinger, 2021-08-13:
Thanks! I'm embarrassed to admit that I hadn't seen that. Must be my
way-past-middle-aged eyes.

lukasmatena, 2021-09-01:
Individual brims are implemented in 2.4.0-alpha1. Closing.

closed 2021-09-01 by lukasmatena

wnovinger, 2021-09-01:
Thanks for the update. 

☸️Sent from my iPhone 11 Pro Max☸️