prusa3d/PrusaSlicer #2456: Configuration import failed

GuillaumeAntoine, 2019-06-05:
### Version
PrusaSlicer 2.0.0

### Operating system type + version
macOS 10.14

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3s with the last firmware

### Behavior
* _Describe the problem_
When you try to import a configuration from a gcode file, you get an error. 

* _Steps needed to reproduce the problem_
Generate a gcode
Close PrusaSlicer
Open PrusaSlicer
Choose File / Import / Configuration import (not sure about the label because it is in French — another issue because you can't choose the appropriate language) or push cmd + L
Choose the gcode file and validate your choice.

* _Expected Results_
The configuration is loaded and saved into the preset configuration

* _Actual Results_
The configuration can't be loaded

[image]

If you edit the gcode and replace the first line with the one from a previous version (ie slic3r PE), that's work.

GuillaumeAntoine, 2019-06-06:
If you replace the first line in the gcode generated by PrusaSliser like this one : 
`; generated by PrusaSlicer 2.0.0+ on 2019-06-06 at 10:36:05`
By the first line from a gcode generated by Slic3r Prusa Edition like this one : 
`; generated by Slic3r Prusa Edition 1.41.3+ on 2019-05-27 at 19:46:56`
it's working.

Migamix, 2019-06-15:
this also occors on every windows version that can run prusa slicer. 
import requirement still reflects old name, it can care less about the version number.

bubnikv, 2019-06-17:
it has been fixed on the master, the fix will be contained in the next
release.

GuillaumeAntoine, 2019-06-17:
Great news. 
Thanks for the update.

closed 2019-06-17 by GuillaumeAntoine