prusa3d/PrusaSlicer #2457: Can't choose the language

GuillaumeAntoine, 2019-06-05:
### Version
PrusaSlicer 2.0.0

### Operating system type + version
MacOS 10.14

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3s with the last firmware

### Behavior
* _Describe the problem_
The software is in French as my operating system

* _Steps needed to reproduce the problem_
Install PrusaSlicer on a computer with the OS in French

* _Expected Results_
Need to change the language of PrusaSlicer in English

* _Actual Results_
No possibility to change the language

odx, 2019-06-10:
Yes. The translations are really lousy and all documentation an discussions on the internet are in English...

GuillaumeAntoine, 2019-06-17:
I don't think it's really lousy but as you said, all documentation and discussions are in English so I prefer to get it in English as on the printer.

bubnikv, 2019-06-17:
The translations were done by native speakers and our customers, so they
should be sound both technically and linguistically.

You may change application language from the "Configuration" menu.

GuillaumeAntoine, 2019-06-17:
Hi bubnikv, 

I didn't find this menu. 
It's all good now. 
Thank you. 

I close it.

closed 2019-06-17 by GuillaumeAntoine