prusa3d/PrusaSlicer #2458: [2.0.0x64] It never ends slicing any model, and crashes if trying to abort operation

Liger0, 2019-06-05:
### Version
2.0.0 x64

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
Custom printer (cr10s), Marlin FW

### Behavior
If I try to slice anything (even very little parts), it never ends slicing. 
If I try to abort, the software crashes.
"Avoid crossing perimeters" is already disabled.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
[Slicing.zip](https://github.com/prusa3d/PrusaSlicer/files/3258903/Slicing.zip)

jasontibbitts, 2019-06-05:
Loaded into 2.0.0 running on my Fedora 29 machine.  Renders immediately and didn't take too long to slice, given that I'm doing a parallel build of PrusaSlicer in the background.  My CPU is, however, fairly performant (an i9-7900x) and I have 64GB of memory.  I was also able to click the cancel button during slicing without error.

Liger0, 2019-06-05:
Ryzen 5 1600 and 16gb of ram.

Liger0, 2019-06-13:
> Loaded into 2.0.0 running on my Fedora 29 machine. Renders immediately and didn't take too long to slice, given that I'm doing a parallel build of PrusaSlicer in the background. My CPU is, however, [...]

I did a video. Sometimes it even ends slicing while the bar still slices. Anyway it crashes when trying to abort.

https://www.youtube.com/watch?v=2avigTEA3Kc

SnorreSelmer, 2019-06-14:
I have some models that take upwards of 2hrs to slice. The model you sent took about a minute to slice.

Windows 10 x64 (1903)
Intel i7 7700
32GB RAM

Liger0, 2019-06-14:
The problem here seems that the slicing ends (I can use the preview!!) but never seems to acknoweledge it, so it crashes if you try to abort the (already ended) slicing.

bubnikv, 2019-06-14:
My best bet is that you are low on GPU memory and for some reason the graphics driver did not play well with slicer. Does it happen for every model? Does it happen for smaller models?

Liger0, 2019-06-14:
It happens on all models. I use a vega 56 and it was idle..

bubnikv, 2019-06-14:
You may try to update your graphics driver then.

Liger0, 2019-06-14:
They are updated.

bubnikv, 2019-06-14:
That's bad.

There is not much we can do without getting our hands onto your machine. There are methods to help, but they are quite involved, like to compile and run slicer under the debugger, to use an OpenGL API tracer as https://github.com/apitrace/apitrace .

We may produce debugging builds of PrusaSlicer one day that would log all OpenGL calls, but we don't have that as of now.

Liger0, 2019-06-14:
Yeah that seems hard for me. I will wait and hopefully one day the dubigging build will be a thing.

Liger0, 2019-06-16:
Just a screnshot to show the hardware utilization while the slicer tries to slice and never ends.
[image]

sethpark777, 2019-07-25:
I am experiencing the same issues. I am on 2.0.0, win10, 64 bit. 

I also have a ryzen 5 1600, 16GB RAM, GTX1070 with updated drivers.

Some models seem to slice fine, but then others hang like this. CPU and GPU use never seem to exceed 10%.

bubnikv, 2019-07-26:
@sethpark777 what operating system? Did you compile slicer yourself?

sethpark777, 2019-07-26:
@bubnikv Windows 10 64 bit, not compiling the slicer myself but that could be something i attempt.

sethpark777, 2019-07-27:
It seems to be the "avoid crossing perimeters" settings that is causing the issue. When disabled, slicing takes about 6 seconds.

Liger0, 2019-07-27:
Even when slic3r worked by itself in old releases, the "avoid crossing perimeter" option, if enabled, always made my slic3r crash.

bubnikv, 2019-08-25:
Aborting issue of the "avoid crossing perimeters" feature has been fixed with 3addae3, and the feature is now also significantly faster.

The model provide takes 75 seconds now to generate the G-code with the "avoid crossing perimeters" feature enabled on my computer, 50 seconds with the "avoid crossing perimeters" feature disabled. The print time is 15h50m with "avoid crossing perimeters" enabled, 15h46m with "avoid crossing perimeters" disabled. 

Closing.

closed 2019-08-25 by bubnikv