prusa3d/PrusaSlicer #2459: Crash when custom symbol in object name

gleibfried, 2019-06-05:
### Version

PrusaSlicer 2.0.0

### Operating system type + version

Windows 10

### 3D printer brand / version + firmware version (if known)

Prusa i3 MK3 with the last firmware

### Behavior
* _Describe the problem_
PrusaSlicer crashes when there are some special characters in the object name

* _Steps needed to reproduce the problem_
When exporting my object directly from Fusion 360 to PrusaSlicer, the object name is the path of the temporary object generated by Fusion and it contains custom symbols.
If I by mistake enter the name edition but don't change anything, I get the error "The supplied name is not valid; the following characters are not allowed: <>:/\|?*" and then it crashes.
Since it's a path we have characters like : and /.

* _Expected Results_
Not completely crash if this error happens.

* _Actual Results_
Crash

alexzkter, 2019-07-13:
Thanks for reporting this. I'm experiencing this as well.
A workaround is to save your file on Fusion360 before sending it to PrusaSlicer.

YuSanka, 2019-08-06:
Thanks for your report. But we can't reproduce this bug. Could you be more specific or at least send us exported file.

YuSanka, 2019-08-06:
I think, I can reproduce your issue. Issue appears when you try to rename object and accidentally move mouse to the Scene. Am I right?

gleibfried, 2019-08-06:
Hi, yes it's indeed when going in renaming and exiting it without changing,
it probably still tries to rename with the generated temporary name and
might have forbidden characters.
I can test again later when I'll have access to my laptop if you need to.

YuSanka, 2019-08-06:
In any case, it looks like this bug is fixed in https://github.com/prusa3d/PrusaSlicer/commit/666e9687ce16e386d218692d01eef6979fb9bd69 and will be a part of upcoming release.

closed 2019-08-06 by YuSanka