prusa3d/PrusaSlicer #2460: Certain GCode lines make print not work

Coffee-Cream, 2019-06-05:
### Version
PrusaSlicer Version 2.0.0

### Operating system type + version
Windows 1

### 3D printer brand / version + firmware version (if known)
RepRap Guru Prusa i3 MK2 with Marlin
Running on OctoPrint on a Raspberry Pi

### Behavior
Recently, I started using PrusaSlicer rather than what I was using before, Slic3r 1.3.0. Whenever I loaded and try to print a model that was sliced with PrusaSlicer, the Z axis motors make a humming sound and don't move. This was not an issue when I used Slic3r. I did some research and testing and compared the GCode files, and I found out that at the beginning of the GCode, there are some lines that I have highlighted that appear from the PrusaSlicer GCode but not the Slic3r gcode. When I commented out those lines, it worked. I was wondering if there is a setting to make PrusaSlicer not generate these lines. Thank you!

These screenshots are the GCode of the same model from PrusaSlicer and Slic3r.

PrusaSlicer:  (The lines that made the print not work are highlighted)
[image]

Slic3r:
[image]

lukasmatena, 2019-07-22:
The lines you have highlighted set maximum accelerations etc. and PrusaSlicer only emits them when Printer Settings->GCode flavor is set to Marlin. Marlin firmware should support them. Either you are using some old Marlin version that does not support the gcodes (I don't know when they were introduced) or there is something else wrong. Does it work when you're not using OctoPrint?

Either way, setting the gcode flavor to RepRap will prevent PrusaSlicer to write that to the output gcode. But I'd like to know what exactly is wrong.

Coffee-Cream, 2019-07-23:
Thanks for the reply luksametena. I tried using a gcode file generated by PrusaSlicer using the same settings I had with the post using Repetier instead of OctoPrint and the issue where the Z Axis motor wouldn't move was still there. I believe that it is probably an issue with my printer, since those GCode command didn't work with OctoPrint or Repetier, even though my printer's documentation said that it supported Marlin firmware.

lukasmatena, 2019-07-23:
Ok, so it is apparently not PrusaSlicer's fault. Is it ok to close this issue then?

Coffee-Cream, 2019-10-06:
Yeah, thanks for your help.

closed 2019-10-06 by Coffee-Cream