prusa3d/PrusaSlicer #2461: Single test suite failure on two minority architectures

jasontibbitts, 2019-06-05:
### Version
2.0.0

### Operating system type + version
Fedora 31

I'm working on getting this package imported into Fedora.  By default, packages in Fedora build for all of our supported architectures, which as of Fedora 31 consists of 64 bit Intel, 64 bit ARM (aarch64), 64bit PPC (little-endian), 64bit IBM system Z (s390x), 32bit Intel, 32bit ARMv7.  Of course, we aren't often able to do proper testing of the software on some of these, but we do run included test suites, and that's what I've done in my prusa-slicer packaging.

Unfortunately exactly one test suite fails on both 64bit ARM and s390x, and while it could be argued that nobody would ever run this software on an IBM mainframe, it isn't beyond reality to think that it might be run on some 64bit ARM machine somewhere.  Plus these odd architectures serve as useful test cases for architectural assumptions.

The failing test is in t/03_point.t and the failing test code is simply this:
```
    my $line = Slic3r::Line->new([50,50], [125,-25]);
    is +Slic3r::Point->new(100,0)->distance_to_line($line),  0, 'distance_to_line()';
```
The test output is:
```
2: #   Failed test 'distance_to_line()'
2: #   at t/03_point.t line 47.
2: #          got: '3.92523114670944e-15'
2: #     expected: '0'
2: # Looks like you failed 1 test of 24.
```
Note that the specific failing value is identical on both aarch64 and s390x.

The only thing I can guess is that the hardware floating point implementation must be subtly different on those platforms in a way that gives a very small variation.  Without an epsilon value the comparison fails.  Obviously plenty of other similar tests in the suite pass without issues; the only problematic one is this one.  Since this particular test stands alone from other similar tests, I wonder if maybe this is testing some specific case which was seen in the past.

I could obviously just disable that one test on those platforms, patch in an epsilon value, or simply exclude those architectures, but I'd rather understand more about what's happening.

bubnikv, 2019-08-07:
I think that the 'is' operator of Perl unit tests compares strings, therefore the result is first converted to a string and then compared to the expected value. Otherwise I would not be able to explain that all the floating point compares can pass. There is an implicit rounding inside the double to string conversion, that's why one does not need the epsilon arithmetic.

To me, it looks like there may be issues with the Perl implementation if it behaves differently on these two architectures.

I am not sure how to handle this particular issue. I would say just remove the line (or I can do it in our repo). The single line does not bring much.

jasontibbitts, 2019-08-07:
For the record, Fedora is carrying the following patch, applied only on those two architectures.  I can try to find a Perl expert who might be able to figure out why only this one result seems to differ there.  I would suspect that somehow very slight differences in floating point representations result in just enough accumulated error to result in a failed comparison.  I don;t know why those particular values were chosen for the tests; they came in with commit d8be67c28bca470bcc0cb334d3073d122c296404.  Perhaps they were chosen to find problems like this.
```
diff --git a/xs/t/03_point.t b/xs/t/03_point.t
index cb71f68..0cc58e1 100644
--- a/xs/t/03_point.t
+++ b/xs/t/03_point.t
@@ -44,7 +44,7 @@ ok !$point->coincides_with($point2), 'coincides_with';
 
 {
     my $line = Slic3r::Line->new([50,50], [125,-25]);
-    is +Slic3r::Point->new(100,0)->distance_to_line($line),  0, 'distance_to_line()';
+    cmp_ok(abs(Slic3r::Point->new(100,0)->distance_to_line($line)), '<=',  4e-15, 'distance_to_line()');
 }
 
 {
```

bubnikv, 2019-08-08:
I have applied your fix with b7f93292fa147d9f6bfd0ca7ae7df03728fe0c7b
Thanks for your time.

I really think that this issue is purely a coincidence, and that one should use the epsilon test for floating point compares in unit tests nearly everywhere (there may be exceptions where an exact floating point value should be returned, but this is clearly not the case).

We will get rid of the Perl tests one day, and the C++ tests will have to use the epsilon tests.

closed 2019-08-08 by bubnikv