prusa3d/PrusaSlicer #2465: [2.0] Arrange arranges wipe tower

Goury, 2019-06-06:
### Version
2.0

### Operating system type + version
Ubuntu

### Behavior
* Arrange arranges wipe tower even when no wipe tower is required (multitool machine but only one tool is used).
* But if I click Slice now then wipe tower disappears.
* Another click on Arrange icon now will rearrange stuff without considering wipe tower (that is gone).
* This is confusing, counter-intuitive and undesirable for me.

Older Slic3r never arranged wipe tower for me.

### How it should be
If you really want us be able to arrange this wipe tower on the plate, it should be configurable if arrange icon would arrange it or not.
Also it would be nice if wipe tower didn't exist on empty plate even for multitool machine with wipe tower enabled and only been added if it's really required.
It shouldn't be that hard, it's the simplest condition: only add this tower if more than one tools is used, otherwise don't add it.
There's no need to preslice the plate to make such decision.

lukasmatena, 2019-06-06:
> Older Slic3r never arranged wipe tower for me.

Yeah, and as you know, there were feature requests https://github.com/prusa3d/PrusaSlicer/issues/2465 and https://github.com/prusa3d/PrusaSlicer/issues/1485 to change that behavior. Which we did, because it seems like a logical thing to do. To me at least. The original behavior was not even intended, the wipe tower just wasn't implemented in the first version of arrange.

> Also it would be nice if wipe tower existed only if it's really required

It would, but it may not be so easy to do. There can be unprintable volumes, modifier meshes, objects with wipe into infill enabled, object without infill with infill set to some specific extruder and so on. That's why this decision is postponed until after the slicing is done. Also, you would still not know the dimensions of the tower before it's actually generated.

> It shouldn't be that hard

Many feature requesters claim that under various circumstances.

I don't really care whether this is or isn't changed to previous state. It is really not about what we "want" you to do. But I'd like to see some consensus first. If we were to vote, I'm happy with how it currently is.

lukasmatena, 2022-04-21:
I think I will close this now and keep current behavior. Nobody else has reported it as a problem in almost three years, so I believe that it fine for most people. Sorry.

closed 2022-04-21 by lukasmatena