prusa3d/PrusaSlicer #2466: PrusaSlicer Error - The software does not start any more

jokohoko, 2019-06-06:
### Version
2.0.0

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
Stock MK3's

### Behavior
The software does no longer start. It shows the errors: 
`Internal error: <unspecified file>(14):'=' character not found in line`
and:
`Unhandled unknown exception; terminating the application.`

I found a similar issue (#2438) report from a few days ago that was closed with the following details:

> Looks like something got corrupted in my AppData directory. Deleting the prusaslicer directory there seems to have done the trick. I am still unsure as to the "why", but if someone else stumbles upon this, try deleting C:\Users\{UserName}\AppData\PrusaSlicer*.
_Originally posted by @Eashley in https://github.com/prusa3d/PrusaSlicer/issues/2438#issuecomment-498125506_

But i have it on 2 individual computers now in the past week, so it seems to be something that happens more often, so a solution would be nice. (or we must have an other common bug on our computers here at the University building). 

[image]
[image]

Also older versions no longer work after this and re-installation does not help either. Only manually removing these files from AppData.

Pille1988, 2019-06-06:
It seems like the update wiped your PrusaSlicer.ini file. What you can do to solve this issue without having to delete your PrusaSlicer folder under Appdata is:
1. create a backup of your existing folder (rename it PrusaSlicer.bak)
2. start Prusa Slicer and close it again
3. copy the content from the PrusaSlicer.ini file in the newly created PrusaSlicer folder to the empty PrusaSlicer.ini file in your PrusaSlicer.bak folder
4. delete the new PrusaSlicer folder and rename the backup folder back to PrusaSlicer

NOTE: Until this issue is fixed, I would recommend keeping a backup of your PrusaSlicer.ini file.

bubnikv, 2019-08-24:
While we have no explanation on how the PrusaSlicer.ini file could have gotten corrupted, we suspect that the Windows API for renaming files is not atomic and it may in some cases lead to file corruption. We replaced the  "file rename" function with a complex function from the [llvm project](https://github.com/llvm) with fd3fe75d1ca4ddf1a36eb89d7210a4ec0e8f4afa. Let's hope it will fix this mysterous issue.
Closing, the fix will be part of the PrusaSlicer 2.1.0-beta.

closed 2019-08-24 by bubnikv