prusa3d/PrusaSlicer #2467: Where is the documentation?

tobymurray, 2019-06-06:
### Version
Version 2.0.0+linux64

### Operating system type + version
Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + MMU2S

### Behavior
I think the old Slic3r PE had a built in link to the Slic3r documentation, now that the projects have parted ways is PrusaSlicer going to be hosting documentation? Even "official" user-contributed documentation would go a long way.

For a practical example, give someone an MMU2S and ask them to figure out how to use one filament for support material, a second for support interface, and a third for the actual print. How do people figure out how to use the program without knowing already?

_Is this a new feature request?_
Yes, if there is no existing documentation

sticilface, 2019-07-05:
I can't find any either.  I'd like to know what some of the properties actually do!

gregsaun, 2019-07-14:
Thank you for this one, hope to see this fixed soon!

bubnikv, 2020-10-17:
Documentation is here
https://help.prusa3d.com/en/article/general-info_1910#_ga=2.180715047.1016148271.1602868785-1539679296.1599661580

closed 2020-10-17 by bubnikv