prusa3d/PrusaSlicer #2468: Provide representative screenshot.

jasontibbitts, 2019-06-06:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

hroncok, 2019-06-06:
licence here is for metadata (the appdata file itself)

hroncok, 2019-06-06:
BTW there is a section about how such screenshot may look like [at the docs](https://www.freedesktop.org/software/appstream/docs/chap-Quickstart.html). Search for `Screenshot size, shape and format`. Copying here for your convenience:

### Screenshot size, shape and format:

- All screenshots should have a 16:9 aspect ratio, and should have a width that is no smaller than 620px (software center applications will be able to fill in the screenshots in the space they provide for that more easily then).
- Ideally the window will be resized to a 16:9 aspect ratio, but screenshots can also be cropped if only a small area of the window needs to be shown.
- Screenshots should be in PNG or JPEG format. PNG is the preferred format; JPEG should only be used when screenshots include large photographs or other images where a lossy format like JPEG may compress better.
- Do not scale screenshots below their original size.

### Basic guidelines for things to include (and not include) in your screenshots:

- Use the default visual and theme settings, including the default font, icons, and window controls. Avoid including your own tweaks to the standard distribution offering.
- Screenshots should be taken with English as the display language.
- Your default screenshot should give an overview of your application, and should show an entire application window. It can be combined with screenshots which show specific aspects of the application.
- Screenshots should not show anything outside the application's windows (including the background wallpaper). If you are taking a screenshot of the whole window, use your screenshot tool's "window" mode (including any window borders in the screenshot, and ensuring that the resulting image has an alpha mask for the rounded corners).
- Some applications, such as games, primarily run full screen. Screenshots of these applications should be taken with the application running full screen - there should be no visible window controls in the screenshot.
- Don't add shadows to your screenshots.
- Do not include content that might be considered offensive or controversial, and avoid showing personal information. Remember that your screenshots will be visible to the internet at large.

### Additional advice on how to take effective screenshots:

- Each of your screenshots should focus on one thing and one thing only. Screenshot one window at a time, and avoid having overlapping windows or user interface elements. This will make it much easier for people to understand what you are showing them.
If a screenshot is demonstrating a single feature or aspect of the application, crop it to exclude irrelevant detail.
- Screenshots often need to feature content, such as images, documents, web pages or videos. Don’t show your application in an ‘empty’ state, and try and use high quality content which has positive associations and broad appeal.
- In general, you shouldn't include the mouse pointer in your screenshots.

hroncok, 2019-06-06:
Alternatively, if we provide the screenshots would you accept them in the repo?

vojtechkral, 2019-06-07:
Fedora doesn't have any location for storing these screenshots?

hroncok, 2019-06-07:
The point is that this is not a screenshot for Fedora, but for any appdata
consumer. It's an universal freedesktop standard. Keeping the screenshots
in upstream is prefered.

jasontibbitts, 2019-07-25:
Let's just do this...

[image]

closed 2019-07-25 by jasontibbitts