prusa3d/PrusaSlicer #2472: Feature Request: Export Gcode Preview Geometry

ardenpm, 2019-06-07:
As this is a feature request, version and printer information is omitted.

The Gcode preview geometry that PrusaSlicer produces is potentially very useful for people who wish to render high quality 3D images that approximates the appearance of a printed object.

To utilise it, an option (definitely for advanced users) to export the Gcode preview geometry as an OBJ file (or other format that retains normals) would be extremely useful. I have hacked this in to a local clone for testing, the results can be scene in this Twitter thread:

[https://twitter.com/ardenpm/status/1136540501073510400](https://twitter.com/ardenpm/status/1136540501073510400)

I was surprised at how decent the results were despite the relatively low detail level in the Gcode preview. There seems to be a positive reaction to this so I think others who have rendering software would appreciate a means to generate similar data. As a starting point, here is the code I hacked in to dump the geometry to the .mi format of the renderer I use at [migenius](https://www.migenius.com). This is in the GLCanvas3D::_load_gcode_extrusion_paths method in src/slic3r/GUI/GLCanvas3D.cpp, replacing the final if statement in that method.

```
    // finalize volumes and sends geometry to gpu
    if (m_volumes.volumes.size() > initial_volumes_count)
    {
        // Output file and some setup
        std::ofstream mifile;
        mifile.open("c:/temp/test.mi");
        int obj_count = 0;
        const char* tab1 = "    ";
        const char* tab2 = "        ";

        for (size_t i = initial_volumes_count; i < m_volumes.volumes.size(); ++i)
        {
            GLVolume* volume = m_volumes.volumes[i];
            
            // For naming
            obj_count++;

            // Object preamble
            mifile << "object \"print" << obj_count << "\"" << std::endl;
            mifile << tab1 << "visible on" << std::endl;
            mifile << tab1 << "tagged off" << std::endl;
            mifile << tab1 << "group" << std::endl;
            mifile << tab2 << "# vectors" << std::endl;

            // Write vectors
            size_t vec_count = volume->indexed_vertex_array.vertices_and_normals_interleaved.size();
            std::vector<float> values = volume->indexed_vertex_array.vertices_and_normals_interleaved;
            for (std::vector<float>::iterator it = values.begin(); it != values.end(); ++it) {
                mifile << tab2;
                mifile << *it << " ";
                it++;
                mifile << *it << " ";
                it++;
                mifile << *it << " ";
                mifile << std::endl;
            }

            // Write vertices
            mifile << std::endl << tab2 << "# vertices" << std::endl;
            for (size_t i = 0; i < (vec_count / 3);  i+=2) {
                size_t v = i + 1;
                size_t n = i;
                mifile << tab2 << "v " << v << " n " << n << std::endl;
            }

            // Write quads
            mifile << std::endl << tab2 << "# quads" << std::endl;
            for (size_t i = 0; i < volume->indexed_vertex_array.quad_indices.size(); i += 4) {
                mifile << tab2 << "p "
                    << volume->indexed_vertex_array.quad_indices.at(i) << " "
                    << volume->indexed_vertex_array.quad_indices.at(i + 1) << " "
                    << volume->indexed_vertex_array.quad_indices.at(i + 2) << " "
                    << volume->indexed_vertex_array.quad_indices.at(i + 3) << std::endl;
            }

            // Write triangles
            mifile << std::endl << tab2 << "# triangles" << std::endl;
            for (size_t i = 0; i < volume->indexed_vertex_array.triangle_indices.size(); i += 3) {
                mifile << tab2 << "p "
                    << volume->indexed_vertex_array.triangle_indices.at(i) << " "
                    << volume->indexed_vertex_array.triangle_indices.at(i + 1) << " "
                    << volume->indexed_vertex_array.triangle_indices.at(i + 2) << std::endl;
            }

            // Object postable
            mifile << tab1 << "end group" << std::endl;
            mifile << "end object" << std::endl << std::endl;

            mifile << "instance \"print" << obj_count << "_inst\" \"print" << obj_count << "\" end instance" << std::endl << std::endl;

            volume->bounding_box = volume->indexed_vertex_array.bounding_box();
            volume->indexed_vertex_array.finalize_geometry(m_use_VBOs && m_initialized);
        }

        // Add the instgroup
        mifile << "instgroup \"geometry_grp\"" << std::endl;
        for (int i = 1; i < obj_count + 1; i++) {
            mifile << tab1 << "\"print" << i << "_inst\"" << std::endl;
        }
        mifile << "end instgroup" << std::endl;

        // Finish up
        mifile.close();
    }
```

This is messy and not optimal but gives an idea of how the geometry can be extracted from the GLVolume and our layout isn't that different to OBJ. There is likely a better technique and place to do it to allow to be triggered on demand instead of during the slicing process however since the slicing display deletes all of the data once uploaded to the GPU it is nessessary to do so prior to the indexed_vertex_array.finalize_geometry() call on the volume.

Just exporting the data to a usable format would be a great start and extremely useful in itself. If that was a popular feature I think the following additional functionality would be helpful, but they would probably be separate feature requests.

* UV coordinates on the extruded geometry, useful for texturing.
* Ability to output a more heavily tessellated preview for smoother results.
* Option to output other data to vertex colours (e.g., speed, temperature).
* Preview Plugin API so renderer vendors could integrate custom preview renderers.

Those are just some crazy additional ideas, I'd stress that just a very basic export to basically make my hack somewhat official would be perfect for now.

fntsmn, 2019-06-07:
Really like this idea, I will love to be able to render easily a object like is printed! Hope they will insert this into PrusaSlicer

bubnikv, 2019-08-22:
Implemented with 1f6aab312b544ff2c0225cc3516c543a3454f078 730283a9e9dc06426b1927e60c1dca52ef2da571 58473f84eedac0556e11ab54db6f58ac0d1353a3 f63b3ba600d26efcd31db87979849d6fc25fc88e 1b6490af4cded5d0fc4a1f25bac430d5aa4a9416
It will be part of the upcoming PrusaSlicer 2.1.0-beta.
Closing.

closed 2019-08-22 by lukasmatena

vertigo235, 2019-08-24:
what is a good renderer to use?

ardenpm, 2019-08-25:
There are lots to choose from, if you are looking for something free though Blender is very capable (and much easier to use since 2.80), however it's a fully fledged content creation tool not just a renderer so it has a bit of a learning curve. Wekster demonstrated in his most recent [video](https://www.youtube.com/watch?v=TnlTJem_gPk) just importing back into Fusion 360 and rendering there if you were already using that for modelling. There is no shortage of options though. In my case, our company writes rendering software using NVIDIA Iray so that is what I had access to.

bubnikv, 2019-08-25:
https://corona-renderer.com/ co-founded by Jarda Krivanek, my university
class mate, maybe Jarda is known to Paul Arden :-)

ardenpm, 2019-08-25:
While I don't know Jarda personally, Corona and of course Chaos Group who now own it are well know to us :-)

Such a small world!