prusa3d/PrusaSlicer #2474: Arrange does not place items efficiently.

ctag, 2019-06-07:
### Version
2.0.0_rc2-1 from Archlinux AUR.

### Operating system type + version
Archlinux

### 3D printer brand / version + firmware version (if known)
i3 MK3S

### Behavior

If I place several items (24 in example) on the plate, and click "Arrange [A]" they will be arranged, but the order of printing each layer involved considerable travel time that makes the print take longer.

To decrease print time, I would expect the items to be laid out such that the hotend moves between neighboors whenever possible to have the shortest travel distance.

[image]

[image]

_Is this a new feature request?_

Possibly. I don't know if it counts as a bug as nothing is broken.

#### Project File

[ir_blinder.zip](https://github.com/prusa3d/PrusaSlicer/files/3265905/ir_blinder.zip)

tamasmeszaros, 2019-06-10:
Hello @ctag,

good point. We will address this as soon as we will have the cycles. Thank you for the feedback.

rtyr, 2020-11-18:
Fixed. Closing.

closed 2020-11-18 by rtyr