prusa3d/PrusaSlicer #2475: New Slicer G-Code Issues with MK3S Display and Initial bead

ghostop14, 2019-06-07:
Hi Prusa Team, I really like the new slicer, but I noticed a couple of small issues after upgrading:
1. When printing new G-Code the printer display clock counts up instead of down.  I looked in settings both on the printer and in the slicer and I don't see any options to change it.  Time remaining to me is more important than elapsed time.

2. It seems the new g-code doesn't automatically lay the initial print area edge bead down anymore.  Personally I found that helpful for consistent filament feeding as soon as the head moved to start the print.

lukasmatena, 2019-07-22:
1. Printer Settings->General->Support remaining times must be checked in order for PrusaSlicer to emit the M73 gcode with the estimated time to finish.

2. The initial print line was always included in the Start G-Code, which is part of our stock printer profiles (Printer Settings->Custom G-Code->Start G-Code).

Are you sure you've been using correct printer profile? Because both issues would be explained if you haven't.

lukasmatena, 2019-11-28:
No response, closing. Can be reopened if needed - enclose a 3MF in that case.

closed 2019-11-28 by lukasmatena