prusa3d/PrusaSlicer #2476: Upper Window border out of screen

Hikari, 2019-06-07:
### Version
2.0.0+

### Operating system type + version
MacOS 10.14.4 (18E226)

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S with  firmware 3.7.1

### Behavior
* Disconnecting the laptop from a large screen and opening the lid afterwards results in the Prusa Slicer Window being out of bounds of the display and cannot be moved around anymore.
[image]

rtyr, 2021-02-01:
Is this issue still valid? Isn't that normal behavior of most of the applications in the scenario like this? (it is pretty normal in Windows).

Hikari, 2021-02-10:
I cannot test this anymore, since I switched to a Linux machine.

foreachthing, 2021-02-10:
FYI: Just press [WINDOWS] + [UP-Arrow] to maximize the window.
Or [WINDOWS] + [RIGHT-Arrow] to put the window to the right-half of the screen.

rtyr, 2021-03-04:
OK, I am closing this issue.

closed 2021-03-04 by rtyr