prusa3d/PrusaSlicer #2477: Numerous "LNK2019: unresolved external symbol" errors when compiling in VS2019

nebbywan, 2019-06-07:
### Version
Master commit 6da1c9813908e3d832d4554506172aea1ebe9f9f

### Operating system type + version
Win 10 x64 v1903
VS2019 v16.1.2
Boost 1.70
CMake v3.15.0-rc

### Behavior
Noticed that commit c9a847a6a609d405d6798e702eaaeabb62f1e539 was for VS2019 support so I figured I'd give it a whirl. After compiling the dependencies using the latest commit and pointing VS2019 to Boost (apparently DCMAKE_PREFIX_PATH doesn't work for this in VS2019?), I got numerous (>300) "LNK2019: unresolved external symbol" errors associated with Boost when compiling the PrusaSlicer project despite all other projects in the solution completing successfully. 

I've tried downloading the binaries directly from Boost, compiling both Release and Debug variants, and re-cloning the repository, yet these errors always crop up. I realize this is new territory, just thought I'd throw it out there in case it was an actual issue and not just me. Here are the outputs from VS2019 and CMake. Let me know if there's anything you'd like me to try or any other info needed.

[Cmake Output.txt](https://github.com/prusa3d/PrusaSlicer/files/3267171/Cmake.Output.txt)
[VS2019 Output.txt](https://github.com/prusa3d/PrusaSlicer/files/3267173/VS2019.Output.txt)

vojtechkral, 2019-06-07:
> apparently DCMAKE_PREFIX_PATH doesn't work for this in VS2019?

I believe it still works. Your CMake log says:

```
1> [CMake] -- CMAKE_PREFIX_PATH: C:\destdir-64-2019\usr\local (from cache or command line)
1> [CMake] -- Found Boost 1.70.0 at C:/destdir-64-2019/usr/local/lib/cmake/Boost-1.70.0
```
And the filesystem libs seem to be found:
```
1> [CMake] -- Found boost_filesystem 1.70.0 at C:/destdir-64-2019/usr/local/lib/cmake/boost_filesystem-1.70.0
1> [CMake] --   libboost_filesystem-vc142-mt-gd-x64-1_70.lib
1> [CMake] --   libboost_filesystem-vc142-mt-x64-1_70.lib
```
I'm not sure what's up with the linking erros.

nebbywan, 2019-06-07:
Sorry, I think I may misunderstand the depth to which DCMAKE_PREFIX_PATH goes. In VS2013, I don't need to do anything special to build the projects, but in VS2019 I had to specify the boost include and lib directories in each project's properties in order for them to be found (otherwise I ended up with a mess of "cannot find boost/XXXX" errors). But you're right, CMake goes off without a hitch (at least when run from VS2019's new integration of it, the command line is another story but seems like a VS problem from what I was reading on various other forums).

nebbywan, 2019-06-13:
Seems to have been fixed by 821ca0e36a4d85d19ca01b404967f53672b5625f. Closing and thank!

closed 2019-06-13 by nebbywan