prusa3d/PrusaSlicer #2480: PrusaSlicer wants to print entire object at "gap fill" speed

jschall, 2019-06-08:
### Version
2.0.0+linux64

### Operating system type + version
Ubuntu

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3s

### Behavior
[image]
With gap fill speed set to 60 mm/s

[image]
With gap fill speed set to 0 mm/s, no other changes.

[image]
Speed configuration

#### Project File (.3MF) where problem occurs
[proj.zip](https://github.com/prusa3d/PrusaSlicer/files/3267962/proj.zip)

jschall, 2019-06-08:
I just realized that setting "gap fill" to 0 doesn't set it to auto - it disables gap fill instead.

But why would it slow down all of the perimeters as well as the gap fill?

Tylon, 2020-06-12:
I have the same problem.

Version

2.2.0+win64
Operating system type + version

Win10 Pro 1909
3D printer brand / version + firmware version (if known)

Prusa i3 mk3s, Firmware 3.9 and before

But let me explain...

I just take https://media.prusaprinters.org/media/prints/171/stls/14798_5fe7ae5c-399b-4f23-acfa-08f9b5b9b017/v33_rewinder_singlezip.amf
Here I delete all but hub_v3.3.stl and center it.

Now I see, printing will be slow (6h), so I turn up the speed to automatic with 15mm/s volumetric flow.
Let me show it
[image]

But still this
[image]

Hm. Ok. Still slow... why? Everything is gapfill speed. So, lets look at the gaps.
[image]

The only gaps are at the top. So, I just ignore gaps (gapspeed 0), then I get what I expected.
[image]

I would say, this behaviour is not as expected. Hope you could fix this. 
Greetings
Tylon

aknezevi-work, 2025-05-05:
Might be related:
I am using Variable Layer Height feature, which is very powerful.
However, having set speeds hard coded as mm/s when using Variable Layer Height is bad because print will either go too fast on some places or too slow when it could be printing faster.
Therefore, I want to use auto speed.
However, it is not consistent with speed giving. It will unnecessarily slow down the print speed.
It might have to do with gap filling, but is more likely, at least by what I saw, to be an issue with Arachne perimeter generator, which is also super great feature to have.
The fact that all these features can't work together properly is a bug. That, or the fact that nowhere is it written to avoid one when using the others.
I have several projects, sometimes Variable Layer Height "solves" the issue and the print is faster, sometimes it makes the issue and the print is slower. Sometimes it makes no difference. Depends on the geometry, when it really should not so much.

[Auto_Speed_3.zip](https://github.com/user-attachments/files/19976094/Auto_Speed_3.zip)
[Auto_Speed_4.zip](https://github.com/user-attachments/files/19976093/Auto_Speed_4.zip)

[Auto_Speed.zip](https://github.com/user-attachments/files/19976106/Auto_Speed.zip)

closed 2026-09-29 by prusaslicer-bot[bot]