prusa3d/PrusaSlicer #2482: Crash while cutting the model

szymonpk, 2019-06-08:
### Version
2.0.0

### Operating system type + version

<details>
  <summary>System info</summary>
PrusaSlicer
Version:   2.0.0
Build:     PrusaSlicer-2.0.0

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.0.18-gentoo x86_64

OpenGL installation
GL version:   3.0 Mesa 19.0.3
Vendor:       Intel Open Source Technology Center
Renderer:     Mesa DRI Intel(R) UHD Graphics 620 (Kabylake GT2) 
GLSL version: 1.30
Installed extensions:

GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_multi_draw_indirect
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
[... 234 lines cut ...]
[12] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[13] wxEvtHandler::TryHereOnly(wxEvent&)
[14] wxEvtHandler::ProcessEventLocally(wxEvent&)
[15] wxEvtHandler::ProcessEvent(wxEvent&)
[16] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[17] g_closure_invoke
[18] g_signal_emit_valist
[19] g_signal_emit
[20] gtk_propagate_event
[21] gtk_main_do_event
[22] g_main_context_dispatch
[23] g_main_loop_run
[24] gtk_main
[25] wxGUIEventLoop::DoRun()
[26] wxEventLoopBase::Run()
[27] wxAppConsoleBase::MainLoop()
[28] wxEntry(int&, wchar_t**)
[29] Slic3r::CLI::run(int, char**)
[30] main
[31] __libc_start_main
[32] _start
</details>

#### Project File (.3MF) where problem occurs
Unfortunately, I cannot share the model. It is against its license :(

BeldrothTheGold, 2019-06-09:
I am seeing the same crash. I can provide a .3MF. 

### Version
version_2.0.0-79-g6da1c9813

### Operating system type + version
Windows 10 1903

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S

### Behavior
Hard crash due to assert when attempting to cut part LCD X.

 - prusa-slicer-console.exe has triggered a breakpoint.
 - GLCanvas3D::render() called recursively

#### Project File (.3MF) where problem occurs

[CutCrash.zip](https://github.com/prusa3d/PrusaSlicer/files/3269292/CutCrash.zip)

enricoturri1966, 2019-06-10:
Working on it -> https://dev.prusa3d.com/browse/SPE-961

FidelCapo, 2019-06-10:
Thank you for report. This issue should be fixed in next release.
Closing.

closed 2019-06-10 by FidelCapo